RoganDawes/P4wnP1_aloa #160: HID scripts not running correctly

Gwerbyn, 2019-08-01:
When I try to run any HID script with P4wnP1 ALOA, it just adds a new job to the job list on the right side and the status is constantly running, it never finishes. I have tried scripts that come with the pi aloa and none work. There are no errors, it just doesn't work. Even this script doesn't work:
```
layout("us");
press("GUI r");
delay(100);
type("test");
```
Here are some pictures:

[image]

[image]

[image]

Any help is highly appreciated!

Gwerbyn, 2019-08-02:
Just figured out that every time I connect to the Pi it triggers an event called HID CONTROLLER ABORTED with the message "Called abort on HID Controller". Im also using a Pi zero w and Im targeting windows 10 newest build.

rienkthegamer, 2019-08-03:
Do you have Keyboard and Mouse enabled in USB settings?

Gwerbyn, 2019-08-03:
@rienkthegamer Yes, I select keyboard and mouse and press deploy every time.

21weberer, 2019-09-16:
Ok, are you using a USB cable for the attack? Or did you solder on a USB head? If you did the head, check the connections with a multimeter, I had the same issue and ended up replacing the head. If you went the cable route, see if the cable has any words on it like Charging only, etc. If you went with the cable, perhaps it went bad.

Gwerbyn, 2019-09-18:
@21weberer i used the cable and i tried like 5 different cables and none of them work. I used those cables before to transfer files from my phone to my pc.

21weberer, 2019-09-20:
Now then, i am assuming you are using the innermost micro-usb port labeled USB, right?

Gwerbyn, 2019-10-28:
@21weberer Yes I'm using the second port from the top (I tried both a few times just in case). Also another thing is that it's not only on my laptop, it doesn't work anywhere so the problem has to be in the device itself.

21weberer, 2019-10-28:
perhaps you have a faulty device, is there perhaps anyway you could run a raspbian build on there and maybe plug a keyboard into the data slot to make sure the pi's port is ok?

jah-ith-ber, 2020-01-14:
I'm having this same issue. Did you come up with a solution @Gwerbyn ?

jah-ith-ber, 2020-01-16:
I was able to get this working after I re-flashed, though it still says the controller was aborted (still works..?).

Admittedly, I also didn't notice in the demo script that the layout was switched from US to DE halfway through. Figured that out when the mouse was working, but the keyboard wasn't. Anyway, happy to be up and running now.

mame82, 2020-02-06:
Closing this (not reproducible on release image).

Re-open if the issue persists

closed 2020-02-06 by mame82

wowkster, 2020-02-22:
Hi, I am also having the same issue. I have tried everything listed above including reflashing more than once and I still can not get it to work. Help is much appreciated!

Gwerbyn, 2020-03-09:
Well the problem still persists, I still haven't been able to get it to work. I reflashed again with quite a few different programs, bough a few new cables to try and still nothing is happening. Therefore I believe the problem is in P4wnP1 A.L.O.A itself.

Thot-Hunter, 2020-05-05:
Does anyone have a fix to this? Just bought an rpi today, and encountered the same error. Any help is appreciated :)

rienkthegamer, 2020-05-05:
I was able to reliably replicate this same exact problem at a friend. I tried a lot. Then I found the issue. If your Micro-USB cable is cheap enough, it doesn't work. My friend had a 'gamebuino' with it, there came a Micro-USB cable that did work. Have you tried using another micro USB cable if it didn't work, for me it didn't work either, at my friend's house, ALMOST ALL of the MicroUSB cables were fairly **** and didn't work.

whackyhack, 2020-05-08:
Let me affirm that whether a USB cable is "power + data" or "power only" has nothing to do with how much it costs.  Not only that, among power+data cables, the most important characteristics, current bearing capability, cannot be predicted by cost.  I have a microcontroller that is power-hungry.  I found that a flat cable bought from Dollar Tree carries more current than many others.  Meanwhile, some cables that come with premium Bluetooth headphones are power-only.  If you buy online, some cables are sold as "charge cable".  No matter their price, these are power-only.  Some are sold as "charge and data".  They can be very cheap, too.  But there are many that has no such designations.  Then, you'll have to take a gamble.

1byeworld, 2020-06-14:
So if everyone is still active on this issue i found something that might help on the github page for the older version of P4wnP1 which didnt have the web ui (if im corect). so basically this user in the older github page talks about same problem in the older version and he figure out a solution (https://github.com/RoganDawes/P4wnP1/issues/110) and the solution is (https://github.com/RoganDawes/P4wnP1/issues/110#issuecomment-331807315) but the problem i have is that this was for the earlier version and i cant locate the file he is telling us to edit.

hope this helps everyone and sameone might be able to find that file in the new a.l.o.a version

Nojdaz, 2020-08-15:
I believe i solved the mystery

The problem i had was resolved by changing the letters in the layout to capital letters 

Instead of layout("us");  Do layout("US");

newpathdev, 2021-11-17:
Using a USB power and data did the trick

electricWah, 2022-05-16:
I am having this issue as well.  I can confirm that the usb cable works, as I have used it with another HID project. The device is not recognized as a keyboard, so the script is stuck running. I have also reflashed multiple times to no avail. What should I try? I am not familiar with usb gadget on kali

Johni12M, 2025-04-16:
Same issue over here!