RoganDawes/P4wnP1_aloa #163: Old P4wnP1 HID Backdoor

HitItAll, 2019-08-12:
This is not exactly an issue but I was wondering if there were any plans to port the old P4wnP1 USB hid backdoor? I understand that there is the new WiFi backdoor but there are some use cases where the HID backdoor is better (long-term deployment). Thank you.

mame82, 2019-08-13:
Only reason for not having it ported yet, is lack of time

NightRang3r, 2020-01-02:
@HitItAll I have put some instructions and the necessary files here:

[https://github.com/NightRang3r/P4wnP1-A.L.O.A.-Payloads/tree/master/RAW%20HID%20Attacks/Backdoor](https://github.com/NightRang3r/P4wnP1-A.L.O.A.-Payloads/tree/master/RAW%20HID%20Attacks/Backdoor)

Hope it helps...