RoganDawes/P4wnP1_aloa #164: P4wnP1_aloa on raspbian ?

Haypepito, 2019-08-20:
Hi,
Is it possible to put P4wnP1_aloa on a raspbian os ? I don't want to use kali :)

Haypepito, 2019-08-20:
it's for use the auto rndis driver

21weberer, 2019-09-16:
You might be able to, if you can clone the github repo and run the setup script, try using the installation method for the P4wnP1 (not ALOA), that might work.

Haypepito, 2019-09-17:
> You might be able to, if you can clone the github repo and run the setup script, try using the installation method for the P4wnP1 (not ALOA), that might work.

Yeah I now but it's broke the wifi connection i don't know why ... (thx for reply)

21weberer, 2019-09-18:
Hmm. Which connection? the Access Point to get to the control page?

Casul51, 2019-09-18:
> You might be able to, if you can clone the github repo and run the setup script, try using the installation method for the P4wnP1 (not ALOA), that might work.

I didn't get it. Are you recommend to just use notALOA .img, or what?

Casul51, 2019-09-18:
> > You might be able to, if you can clone the github repo and run the setup script, try using the installation method for the P4wnP1 (not ALOA), that might work.
> 
> I didn't get it. Are you recommend to just use notALOA .img, or what?

Also is there is the way to make ALOA to raspbian like installing ra4son drivers and build ALOA (I guess that shouldn't work like that but why not?)

Haypepito, 2019-09-19:
> Hmm. Which connection? the Access Point to get to the control page?
No i can't surf or download something on the web. I want to use p4wnp1 aloa or p4wnp1 just for the auto rndis driver .

21weberer, 2019-09-19:
> > You might be able to, if you can clone the github repo and run the setup script, try using the installation method for the P4wnP1 (not ALOA), that might work.
> 
> I didn't get it. Are you recommend to just use notALOA .img, or what?

No, what I mean is to use the installation method for the original, clone the github, install it, etc. but now that I think about it, I read in other posts that you can't use apt-get upgrade-dist, as it changes the kernel. So if it isent the specific kali image, It probably won't work.