RoganDawes/P4wnP1_aloa #166: Sending data back to P4wmP1

flyer-machine, 2019-08-28:
Did anyone ever tried to send data back to the P4wnP1?
For example, I am retrieving the hostname of a windows target system and want that hostname to be sent back to the P4wnP1 to display it on a little display.
Or I want to switch the payload according to the detected OS-Version on the windows target system.
Or I want to process the retrieved data or the output of a command, and so forth...

I already tried to mount the binary image file via etc/fstab/ which works in the first place.
But as the HID-Script writes a file in this mass storage (on the windows target system), it is not updated/recognized on the kali linux system of P4wnP1.
This was my favorite approach, but as the mounted image file (*.bin) is not in sync it ist quite useless.
I guess it's a linux issue but unfortunately I am not a linux expert.

Is it possible to send data via HID script back to the P4wnP1?
Or is it better to send data via Wifi as it is a PI Zero W?

Any ideas/recommendations are welcome.

Regards, Thomas

1989gironimo, 2019-09-01:
Maybe you can try something like [this](https://github.com/whid-injector/WHID/issues/47)?

flyer-machine, 2019-09-02:
Thank you for this hint!
By now unmounting and mounting the mass storage file seems to work for me.
The generated files are up to date and readable within kali.

closed 2019-09-02 by flyer-machine