RoganDawes/P4wnP1_aloa #169: Portable power for P4wnP1

21weberer, 2019-09-18:
So I had an idea the other day, and that was if i attached a powerboost 500c charger to the pi0 to allow it to power on to preconfigure scripts. it would be something like this: the battery is connected to the powerboost. when the usb is plugged in, the power lines are connected to the input of the powerboost instead, and there would be a switch to tie the ENable pin to ground to turn it on/off. by turning it on, the output would go to the powerlines of the Pi, allowing for it to run. would this work?

whackyhack, 2019-09-22:
Is it correct that "Powerboost" you mention is a converter to 5V with a USB output?  In that case, simply plug into the pi0w's power-only port (the fake mini-USB close to the end).  No need to tap pi0w's USB.

21weberer, 2019-09-24:
the idea was the usb power would provide power input to the powerboost to be charging the battery. yes, that is the board i was talking about, https://www.adafruit.com/product/1944, then the output would be soldered onto the test pads.

closed 2019-10-11 by 21weberer