RoganDawes/P4wnP1_aloa #171: Big Bag

Mrreadiness, 2019-09-22:
Hi, I tried using P4wnP1 and P4wnP1 aloa, and discovered bag in last versions in Windows 10, when I use any functions with HID keyboard, Windows freezes, this is not windows protection, because other bad usb working. Are you know about this problem? If so, can you tell when this will be fixed?
P.S. sorry for my English

21weberer, 2019-09-29:
I had an issue with Legacy P4wnP1 (the old one) where the lockpicker payload wouldent work. I currently don't have my windows pc with me, but I can try to test some stuff out and see if I can get something to work.

MrCook1es, 2020-06-16:
same problem on windows10 when I plug into my laptop, it freeze my windows after 15 seconds (when aloa boots up).
Aloa works fine, I can connect with my mobile phone on wi-fi

thecoder-001, 2020-06-16:
Try to disable all the antivirus suites on your pc (yup, even the windows defender) and then try again. It might be that your antivirus identifies the pi as rubber ducky and tries to protect the computer but causes damage itself. 
Someone had the same issue: #209

MrCook1es, 2020-06-18:
Hi, @thecoder-001 I have disabled the protection of Kaspersky and also of Windows Defender.
I insert the stick, as soon as the system finishes loading and the "P4wnP1" wi-fi becomes visible, the PC freezes.
At that point I connect with the smartphone via wi-fi, disable "HID Keyboard" and the PC starts working again.
I also tried to modify all the USB information (vendor id, product id, manufacter, product, serial) but the problem does not change.
The PC becomes usable as soon as I disable "HID Keyboard" from my smartphone.

thecoder-001, 2020-06-29:
Hello @MrCook1es  , sorry for the late response but I got busy for a few days in school work. Can you try to get logs related to device initialization of USB peripherals when the pi is plugged in with HID keyboard on? I don't know how to do it on windows but there must be surely logging somewhere. Also, can you crank up your PC's volume and notice if theirs any chime (the kind of sound windows use as alert when a device is initialized and ready to use) when you plug in the pi. Also it would be helpful if you can report back the Windows 10 version your PC is running.

vvntrmvte, 2020-07-01:
I am also having this issue, when I do fresh install and hook the aloa device to PC it chimes like a half chime and freezes.  nothing can be done, no keyboard, no mouse.   It's very weird, the aloa device is working fine though, because it's broadcasting the wifi.  I also have issue where it never installs as an Ethernet device... so I can never ssh into it through that means.

thecoder-001, 2020-07-02:
Wow, this seems to be affecting a lot of users lately. Maybe the driver that windows 10 uses to communicate to HID devices (rpi 0w here specifically) is itself bugged in the latest versions. If that's the case then I would soon perform a test by deleting my old pi drivers on my pc and letting windows to install them again. 
Or it can also be the case, though highly unlikely (I can't even why it could be happening suddenly), that the pi is somehow interfering with the device initialization process and resulting in the windows hanging (the device initialization process on windows is very susceptible to failures in case of odd behaviors from the device and can cause wierd effects, like the complete system hanging up.
Also, can anyone of you run any background program (like a video or music, rendering, etc) before connecting the pi and then  connecting it while the process is still running? It can help us identify whether windows is completely hanging or only the external interface becomes unresponsive.

vvntrmvte, 2020-07-02:
Yeah I can do a test for that. I'll report back with my findings. Thanks for your help!

Mrreadiness, 2020-07-02:
I don’t know about music or video, but when I try to use a keyboard or mouse it’s freezing, but after unblocking all what I had tried to press executing, like win or other shortcuts.

MrCook1es, 2020-07-02:
> Hello @MrCook1es , sorry for the late response but I got busy for a few days in school work. Can you try to get logs related to device initialization of USB peripherals when the pi is plugged in [...]

Hi, @thecoder-001  when I insert ALOA, I hear the windows sound of the new peripheral identified and at the bottom left I had time to clear the message "new peripheral detected and installed".

OS: Windows 10 Pro
Version: 1909
Build: 18363.900

when I insert ALOA, I hear the sound of the new peripheral found and at the bottom left I see the message "new peripheral detected and installed".
If I keep a youtube video open, the video freezes like everything else, but the audio continues to move forward correctly.

thecoder-001, 2020-07-03:
Thanks for the info, I am looking into the symptoms and trying to find the issue. Meanwhile, can you give me a screenshot of your USB gadget settings as they are currently? That would help me to replicate the same conditions. By my initial research, it looks like windows particularly dislikes bad storage devices (I can't confirm this fully). Can you try to change your config according to as I have shown below and report back the results?
(match the box ones first, and if it still doesn't work, disable the arrow ones too) [it looks like the covert channel Is yet not ported fully as from to do list: `Porting the full HID covert channel functionality to Go core (I'm on my own with that)`, so it might be causing problem, though I am not sure]
`
[image]

MrCook1es, 2020-07-03:
@thecoder-001 
I did a test with the configuration indicated by you in the figure. Nothing changes, everything freezes in the same way as before.
The only item that makes everything work immediately is to disable "Keyboard - HID keyboard functionality".

Vendor ID
Product ID
Manufacter Name
Product Name
Serial Number
previously I had edited all the items without solving any problems.
At the moment after the reset, I have the default entries.

MrCook1es, 2020-07-03:
now I have redone all the tests obtaining the same results as before.
I confirm that all services work, (including ethernet connection via USB and mass storage), the only service that creates problems is "Keyboard - HID keyboard functionality"

[image]

Mrreadiness, 2020-07-04:
My tests showed the same results.  I have devices like Bad Usb, but there are no such problems there.

thecoder-001, 2020-07-04:
It looks like that a specific part of the Windows is failing, just like @Mrreadiness pointed out. The best way now to diagnose this problem would be to inspect the logs. However, I am not able to replicate the issue with my pi even after multiple reflashing and resetting of the drivers. It also leads me to the belief that some specific generations of onboard USB hubs might be causing the problem as it is not widespread.
Can anyone of you get the relevant logs from your PC (I don't know where the logs are stored for external devices, if I find the location, i would post an update)?
Also can you try to run aloa on another windows machine and check if the issue happens there too? If not, post your machine's specs. This would significantly help us to narrow down to the problem.

Mrreadiness, 2020-07-04:
I’m sure it’s not a Usb hub, I tried about 6. I had a version that the problem was in unlicensed windows, but I also tried on my laptop with Windows preinstalled, it also happens periodically.  What if the problem is in the Raspberry Pi?  At the moment I can not help with logs.  Sory, sometimes I use google translate.

vvntrmvte, 2020-07-04:
So I on a whim, disabled Kaspersky and tried on fresh install and everything worked fine.  No freezing whatsoever.  I was able to login to the web settings via browser or ssh no problem, Everyone having an issue, try turning off any antivirus you may have and plug in the ALOA. Thanks @MrCook1es for your help and being patient.

thecoder-001, 2020-07-06:
No worries @Mrreadiness . Fortunately, we are able to understand what you say and hopefully you are able to understand us too. If you find a sentence difficult and are not able to understand it, I would try to simplify it. Also no problem if you are not able to get the logs, someone else or I can extract them from our systems. Also can you perform the steps in <a href="https://www.howtogeek.com/222730/how-to-find-out-why-your-windows-pc-crashed-or-froze/"> this article </a> (it is the best guide I could find and is from a respectable source).

Also, you told me that you speak a different language natively. Can you and @MrCook1es confirm if anyone of used the P4wnP1_aloa on a non-english windows install or a system which had a different language keyboard attached? If that is the case, then it is a simple and easy fix.

Also, I am happy that @vvntrmvte was able to sort out the issue with his P4wnP1_aloa. As I had mentioned in the starting of discussion, multiple people seem to have problems with antivirus software blocking P4wnP1_aloa. Thanks @MrCook1es for helping him.

MrCook1es, 2020-07-06:
Hi @thecoder-001 I confirm that I am using a non-English version of Windows. And an Italian version with original license.
Thanks for the link you provided to collect the log information, I will try to follow the procedure.

Mrreadiness, 2020-07-06:
Thank you very much for your understanding.  I use Windows 10 for RU region, the default is EN keyboard.

thecoder-001, 2020-07-06:
Hello @MrCook1es and @Mrreadiness , I have formed a basic idea. It looks like the different language layout on your computers and  P4wnP1_aloa is causing things to mess up. Can you try to change your keyboard language (first from GUI, and if it doesn't work, then from CLI)? The instructions are documented <a href="https://github.com/RoganDawes/P4wnP1_aloa/tree/master#keyboard-layout"> here </a>. MrCook1es can try the Italian (IT) layout and Mrreadiness (even if you use EN keyboard, it is worth a try) can you try the Russian (RU) layout?

 Connect it first and then change the keyboard layout (from GUI or CLI), don't worry if the system freezes up and try to run any demo script. If the system is still unresponsive, disconnect the pi / or disable the HID keyboard and enable it again. Then report back the results.
Thanks a lot for your cooperation.

Mrreadiness, 2020-07-06:
Hi again! I tried 'layout("us")' and 'layout("ru")', but it doesn't work. But I recalled that I found an intermediate solution to the problem, it sometimes helps (8 out of 10). "waitLED(ANY_OR_NONE, 5000);" May help solve the problem.

thecoder-001, 2020-07-09:
Sorry I got busy in school work for a few days. So unfortunately changing the layout doesn't work. Can you describe how "waitLED(ANY_OR_NONE, 5000)" is able to help? What happens when it is used in the script? 
And as you were saying, it doesn't work all the times which can mean that windows might need more time to set up the drivers for communication to pi.

Mrreadiness, 2020-07-10:
Hi @ thecoder-001! I started using it a year ago, and now I don’t know where I found it. And I don't know why this works. I used this at startup. I think sometimes this doesn't work because raspberries aren't ready (when this happens, I try to run this command myself).

Mrreadiness, 2020-07-25:
Hi @thecoder-001 and @MrCook1es! Do you have any successes?

MrCook1es, 2020-07-25:
I'm sorry but these weeks are hellish at work

thecoder-001, 2020-07-27:
Hello Mrreadiness,
Unfortunately I am also busy in my school work as my exams have started.But, I also have started reading all the code slowly whenever I find time during my exam preparations so that I can trace out the problem if its due to P4wnP1 malfunctioning.

Mrreadiness, 2020-07-27:
Okay, good luck guys!

thecoder-001, 2020-08-07:
`Status update:` Still haven't been able to discover what's causing it in the little spare time I am getting (still busy in school work). But as more and more people are experiencing the same thing (like issue #247), it can be due to faulty drivers or bug in the Windows external device management stack. Also, @Mrreadiness can you please explain again how "waitLED(ANY_OR_NONE, 5000)" improves the situation? Does it make Windows unfreeze again?
Thanks.

Mrreadiness, 2020-08-07:
Hi! I'm not sure, but I think core of this problem is related to conflicts at the time of loading (the driver is loaded too quickly or something else). waitLED (ANY_OR_NONE, 5000) - stops the Raspberry for a short time.

thecoder-001, 2020-08-07:
@Mrreadiness can you please try to replace "waitLED(ANY_OR_NONE, 5000)" with "waitLED(ANY_OR_NONE, 10000)" and test again? It would give Windows more time before timing out (5 seconds extra) and maybe resolve the problem of things loading up too quickly.

Mrreadiness, 2020-08-08:
Hi! I got something new! After a lot of  tests I understand that num in waitLED is not mature, but if windows freezing and we use that func, windows unfreezing. I don't understand why and how. But I'm sure, this function unblock windows. One of solution to the problem this code:
delay(5000);
waitLED(ANY_OR_NONE, 5000); 
May be we can found answer at source code of waitLED.