RoganDawes/P4wnP1_aloa #172: Not functioning at all

Neuroses, 2019-09-28:
I've been trying for a few days to get P4wnP1 to do anything. Starting the preloaded HID script does nothing, and having it emulate mass storage doesn't do anything either.

21weberer, 2019-09-29:
Using a cable? try a different one that has data lines. Using a USB head you soldered on? try a data cable. Please note to use the innermost port labeled "usb". Try all of that, then tell me if it still dosen't work, I just found my ALOA stick, so I can troubleshoot problems.

Neuroses, 2019-09-29:
I'm trying 3 or 4 different cables at the moment. To make it easy I made P4wnP1 emulate mass storage on startup to show when it's actually functioning, and with 2 of the cables it loaded as a USB drive, but flickered in and out.

21weberer, 2019-09-30:
Flickered in and out? Can you be more specific? Which operating system is the target? i might be able to reproduce the problem.

Neuroses, 2019-09-30:
Windows 10 is my target but I’ve also tried it with Parrot OS on my laptop. I’ve tried 8 different usb cables, and on deploying the usb settings to include emulating mass storage it says there’s an E drive for about half a second, then goes away. I tried out the HID aspect and got it to work just fine.

21weberer, 2019-10-01:
ok, sorry for the delay. i would keep the drive in, then go to disk management/Gparted (I personally reccomend Gparted for linux, not sure if it is preinstalled on parrot) and see if the drive is listed there at all. it could be possible that it needs to be formatted.