RoganDawes/P4wnP1_aloa #175: Problem with web interface (no webserver)

RancidRocket, 2019-10-09:
attempting to connect to the webserver at 172.24.0.1:8000 through firefox yields this error message

An error occurred during a connection to 172.24.0.1:8000. SSL received a record that exceeded the maximum permissible length. Error code: SSL_ERROR_RX_RECORD_TOO_LONG 

attempting to connect to 172.24.0.1:8000 through chrome yields this:
ERR_NETWORK_CHANGED

i have no idea what is causing this but it only happens when connecting to the network for the second time, and can be fixed with a reflash, but that is really time consuming.

21weberer, 2019-10-09:
Is there any setting you changed in the net interface?

RancidRocket, 2019-10-10:
i changed the default wifi profile, and that was it.

21weberer, 2019-10-10:
the default as in what the master template uses? if so, what template are you using?

RancidRocket, 2019-10-11:
Just a custom wifi template, I chsnged name and password, Microsoft edge works every time with this, so I’m not sure if it’s that big of a problem anymore

RancidRocket, 2019-10-11:
Just a custom wifi template, I changed name and password, Microsoft edge works every time with this, so I’m not sure if it’s that big of a problem anymore

closed 2019-10-11 by RancidRocket