RoganDawes/P4wnP1_aloa #177: os detection

learningsomethingnewanduseful, 2019-10-10:
i want to add something like this https://github.com/hak5/bashbunny-payloads/pull/384 to p4wnp1 aloa but i don't really know how. already tried it couple times but cant get it working. this is what i also found from bash bunny. https://github.com/hak5/bashbunny-payloads/blob/master/payloads/extensions/get.sh

21weberer, 2019-10-16:
given that P4wnP1 ALOA is meant to emulate a keyboard, mouse and usb storage device, it might be difficult. I have been thinking about this for some time now, and I know it is not a great method, but to differentiate Windows from OSX, maybe there could be a script monitoring for LED changes, and another could toggle NUM lock, as with HIDScript, you cant use the waitledrepeat code, as it dosen't work with OSX. Probably woulden't help with linux, but it could be a start.

mame82, 2019-10-16:
The bashbunny script linked above utilizes `nmap` OS fingerprinting. 

I can't recommend this for two reasons:

1) It is active TCP/IP communication to the target and thus "loud".
2) Results aren't reliable.

Now if you want to use it anyways, it is pretty easy, because:

- P4wnP1 uses Kali which means `nmap` is right at your hands
- If you start your bash script from a Trigger on DHCP please, there is no need to to do fancy stuff like parsing a lease file for the remote IP, as the bashscript is called with a pre-assigned environment variable holding the target IP (set by the DHCP lease trigger)

SymbianSyMoh, 2020-02-13:
@learningsomethingnewanduseful Check this script out https://github.com/SymbianSyMoh/P4wnP1-ALOA-Payloads/blob/master/pwnpi-OSAutoDetect.sh