RoganDawes/P4wnP1_aloa #178: question

bodri82, 2019-10-15:
hey i might be stupid but how do i transfer files via wifi covert channel ?
I am unable to transfer any file to pi

21weberer, 2019-10-15:
youll need to use ssh. there is no file transfer option via the web client.

bodri82, 2019-10-16:
i already run ssh i mean when i got over c2 agent and have a shell

21weberer, 2019-10-16:
Oh my god im dumb. sorry, I didnt read the question apparently. My recommendation would maybe try using the disk creator script and store files there using xcopy, but there must be a better solution. And kudos to you, I have only got the reverse shell working once for me.

mame82, 2019-10-16:
An example script which consumes the correct environment variables set by different trigger actions could be found here:
https://github.com/mame82/P4wnP1_aloa/blob/master/dist/scripts/trigger-aware.sh

What's left to do is to modify the script to start nmap, place it in a proper TriggerAction for granted DHCP leases, enable CDC ECM (for Mac/Linux) and RNDIS (for Windows).

Once an IP is assigned to the target host's USB interface, the TriggerAction will fire.

If you want to do follow up actions with P4wnP1 from your bash script, the `p4wnp1_cli` command could be used (f.e. to start a HID script, let the LED blink or whatever you like)

mame82, 2019-10-16:
Here is another bash script, showing how to alter P4wnP1 runtime settings using `P4wnP1_cli` instead of using the web frontend.

https://github.com/mame82/P4wnP1_aloa/blob/master/dist/scripts/servicestart.sh

The cli client is very powerful, I highly suggest to use its built in help.