RoganDawes/P4wnP1_aloa #180: No access point is spawned

xyzNOTNOOB, 2019-11-10:
i downloaded and flashed the image from here 
attached the pi to my win10 pc 
searched for the access point that should appear but it didn't 
ssh to the pi on 172.16.0.1 that was ok 
typed dmesg and boom alot of red lines related to wifi driver ,as below
''''''''''''''''''''''''''''
[   27.704683] usbeth: port 1(usb0) entered blocking state
[   27.707038] usbeth: port 1(usb0) entered forwarding state
[   27.723636] IPv6: ADDRCONF(NETDEV_UP): usbeth: link is not ready
[   28.563097] IPv6: ADDRCONF(NETDEV_CHANGE): usbeth: link becomes ready
[   34.263668] brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012
[   34.770197] EXT4-fs (mmcblk0p2): resizing filesystem from 1520130 to 3772982 blocks
[   36.257060] EXT4-fs (mmcblk0p2): resized filesystem to 3772982
[   36.962875] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[   36.970961] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[   36.974714] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[   39.522873] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   39.529764] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
[   42.162847] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   44.722988] brcmfmac: _brcmf_set_multicast_list: Setting mcast_list failed, -110
[   47.282852] brcmfmac: _brcmf_set_multicast_list: Setting allmulti failed, -110
[   49.842854] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[   52.402913] brcmfmac: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
[   60.082855] brcmfmac: _brcmf_set_multicast_list: Setting mcast_list failed, -110
[   62.642863] brcmfmac: _brcmf_set_multicast_list: Setting allmulti failed, -110
[   65.202846] brcmfmac: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
''''''''''''''''''''''''''''
can you help me with this error ?
i googled that before submitting this issue and found that it is driver error and some forums sugessted that i reload it ( modprobe ) but nothing happened.

Subcode, 2019-11-28:
Hi, it seems the repo is very inactive and hasnt been updated in 10 months. 
I'm not sure any work is being done on this anymore.
I havent dived deep into this, so take this with a grain of salt. But i installed this image last week and it worked fine. Could this be a problem with ur usb power? since it says "get_tx_power: error (-110)".
maybe your usb port isnt supplying enough power to run the wifi, i used a externally powered usb hub and had no issues. To test this please connect the rpi to a mobile phone charger or any other charger and see if the wifi shows up, it will be a strange name with a bunch of emoticons.

pquodling, 2019-12-01:
> Hi, it seems the repo is very inactive and hasn't been updated in 10 months.
Yup, appears to be the case. The new Kali version is just out, and it would be handy to know about its integration with this.

sq3xz, 2019-12-06:
Hi,
I have downloaded the same img (hope that is the same;)) from github. Out of box the AP is up and running. But I can't connect to wifi as client and the settings are not stored. After reboot I have the standard settings (out of box settings). I tried to do this from CLI and web (client mode and failover).

xyzNOTNOOB, 2019-12-07:
> maybe your usb port isnt supplying enough power to run the wifi,

i used my phone charger (with 2 amp output) and the same issue.. no access point.

cornoctopus, 2019-12-31:
I can confirm this issue. After flashing sd card there is no AP, and no connection through usb. Cable and power supply are ok, because I can run raspbian with external usb wi-fi card. For usb connection I use usb-board for pi and usb cable.