RoganDawes/P4wnP1_aloa #183: Master Templates do not get loaded after boot

0x00xx, 2019-12-02:
Hello,

I created a new Master Template which simply should not start bluetooth at start up. For that I created such a master template:

https://i.imgur.com/NsdyxUM.png

But it after a reboot Bluetooth is still on. Same happens for usbeth. What am I doing wrong here?

sq3xz, 2019-12-06:
Hi,
are your settings deployed? I have an issue that the settings after reboot are back to defaults.
I change AP settings and after reboot the standard AP is up and running. I can store settings but not deploy.

0x00xx, 2019-12-06:
Hi,
yes, when I deploy them manually the settings are getting deployed. The only thing which does not work is powering bluetooth off with the master template.
Same for the usbeth power switch.

I didn't test it with the AP settings. I overwrote the "startup" template for the AP settings, thus my setting gets loaded there.
This does not work for the bluetooth settings though

sq3xz, 2019-12-06:
I tried your way and added my template
[image: image.png]
After reboot pi have not started. Flashing SD now.

sq3xz, 2019-12-06:
OK.
I selected all templates in master and it is rebooting OK.
But I can confirm BT is always on.