RoganDawes/P4wnP1_aloa #184: ES keyboard not mapped correctly

SsntiCar2, 2019-12-06:
The Spanish keyboard is not mapped corrctly, when the iput is " ' " the computer recognizes it as a ´ tilde, which ruins the sintax in many powershell commands.

SsntiCar2, 2019-12-06:
[image]

mame82, 2020-02-06:
Can't reproduce the issue, this was typed using "es" layout. Assure your HIDScript uses correct ASCII mapping for `'` as P4wnP1 handles UTF-8 and equal looking characters don't necessarily have same Unicode codepoints

mame82, 2020-02-06:
The script used to type out the text above:

```
layout('es');			// Spanish keyboard layout
typingSpeed(0,0)

waitLEDRepeat(NUM);		// Wait till NUM LED of target changes frequently multiple times (doesn't work on OSX)
type("Can't reproduce the issue, this was typed using \"es\" layout. Assure your HIDScript uses correct ASCII mapping for")
type(" `'` as P4wnP1 handles UTF-8 and equal looking characters don't necessarily have same Unicode codepoints")
```

It was typed into the browser on Debian with Spanish (es) keyboard layout selected.

closed 2020-02-06 by mame82

samu-delucas, 2020-12-09:
I have encountered the same issue, although only when using the P4wnP1 in **Windows**. At the moment I use `press("SHIFT 2")` to write `"`, and I still don't know how to write `'`. 

I've found out that this has to do with Windows writting `´` when pressing KEY_APHOSTROPHE and SPACE (`¨` when you also press SHIFT), and Linux writting the correct characters (`' or "`).

HectorHumanes, 2022-08-13:
Same issue here for Spanish keyboard and Windows