RoganDawes/P4wnP1_aloa #187: Can't alter keyboard layout

trifasico, 2020-01-01:
Hi there, im a portuguese coder, and i want to create my layout so that the hid scripts work correctly on my machine, BUT, i have already tried a bunch of stuff and i can't seem to make it work!
I have already tried:
- creating my own pt.json file  and putting it in /usr/local/P4wnp1/keymaps  and in  ~/P4wnp1/dist/keymaps and it doesnt' work
it can't seem to find the fila, so it just uses the lsat layout used
- altering a existing file (ex. it.json and us.json) and it just ignores any alterations i do
- I have already tried to put all the *.json in both folder, with my layout, and it just uses the old layout

So i can't change shit, and i created the pt.json file, but i can't seem to make it work or even try it! Can 
someone please help me?

I upload my pt.json file below, any help contact me! 

[pt.txt](https://github.com/mame82/P4wnP1_aloa/files/4014070/pt.txt)

osiris6457, 2020-01-21:
you have 1 error and 15 duplicates that you dont need. visual studio code shows all errors.

osiris6457, 2020-01-21:
it should be something like this. didn't test it.

{
	"Name": "PT",
	"Description": "Portuguese USB Keyboard report mapping",
	"Mapping": {

            "CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "CONTROL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "LEFT_CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "RIGHT_CTRL": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": []}],
            "ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "LEFT_ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "RIGHT_ALT": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": []}],
            "SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "LEFT_SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "RIGHT_SHIFT": [{"Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": []}],
            "GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "WIN": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "LEFT_GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "RIGHT_GUI": [{"Modifiers": ["MOD_RIGHT_GUI"], "Keys": []}],

            "ESC": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],
            "ESCAPE": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],

            "F1": [{"Modifiers": [], "Keys": ["KEY_F1"]}],
            "F2": [{"Modifiers": [], "Keys": ["KEY_F2"]}],
            "F3": [{"Modifiers": [], "Keys": ["KEY_F3"]}],
            "F4": [{"Modifiers": [], "Keys": ["KEY_F4"]}],
            "F5": [{"Modifiers": [], "Keys": ["KEY_F5"]}],
            "F6": [{"Modifiers": [], "Keys": ["KEY_F6"]}],
            "F7": [{"Modifiers": [], "Keys": ["KEY_F7"]}],
            "F8": [{"Modifiers": [], "Keys": ["KEY_F8"]}],
            "F9": [{"Modifiers": [], "Keys": ["KEY_F9"]}],
            "F10": [{"Modifiers": [], "Keys": ["KEY_F10"]}],
            "F11": [{"Modifiers": [], "Keys": ["KEY_F11"]}],
            "F12": [{"Modifiers": [], "Keys": ["KEY_F12"]}],

            "SYSRQ": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINT": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINTSCR": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "SCROLLLOCK": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "SCROLL": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "PAUSE": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BREAK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BRK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],

            "\\": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
            "1": [{"Modifiers": [],	"Keys": ["KEY_1"]}],
            "2": [{"Modifiers": [],	"Keys": ["KEY_2"]}],
            "3": [{"Modifiers": [],	"Keys": ["KEY_3"]}],
            "4": [{"Modifiers": [],	"Keys": ["KEY_4"]}],
            "5": [{"Modifiers": [],	"Keys": ["KEY_5"]}],
            "6": [{"Modifiers": [],	"Keys": ["KEY_6"]}],
            "7": [{"Modifiers": [],	"Keys": ["KEY_7"]}],
            "8": [{"Modifiers": [],	"Keys": ["KEY_8"]}],
            "9": [{"Modifiers": [],	"Keys": ["KEY_9"]}],
            "0": [{"Modifiers": [],	"Keys": ["KEY_0"]}],
            "'": [{"Modifiers": [],	"Keys": ["KEY_MINUS"]}],
            "«": [{"Modifiers": [], "Keys": ["KEY_EQUAL"]}],

            "|": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
            "!": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_1"]}],
            "\"": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"]}],
            "#": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_3"]}],
            "$": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_4"]}],
            "%": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_5"]}],
            "&": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_6"]}],
            "/": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_7"]}],
            "(": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_8"]}],
[... 121 lines cut ...]
            "UP": [{"Modifiers": [], "Keys": ["KEY_UP"]}],
            "RIGHT_ARROW": [{"Modifiers": [], "Keys": ["KEY_RIGHT"]}],
            "LEFT_ARROW": [{"Modifiers": [], "Keys": ["KEY_LEFT"]}],
            "DOWN_ARROW": [{"Modifiers": [], "Keys": ["KEY_DOWN"]}],
            "UP_ARROW": [{"Modifiers": [], "Keys": ["KEY_UP"]}],

            "COMPOSE": [{"Modifiers": [], "Keys": ["KEY_COMPOSE"]}],

            "NUMLOCK": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],
            "NUM": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],

            "F13": [{"Modifiers": [], "Keys": ["KEY_F13"]}],
            "F14": [{"Modifiers": [], "Keys": ["KEY_F14"]}],
            "F15": [{"Modifiers": [], "Keys": ["KEY_F15"]}],
            "F16": [{"Modifiers": [], "Keys": ["KEY_F16"]}],
            "F17": [{"Modifiers": [], "Keys": ["KEY_F17"]}],
            "F18": [{"Modifiers": [], "Keys": ["KEY_F18"]}],
            "F19": [{"Modifiers": [], "Keys": ["KEY_F19"]}],
            "F20": [{"Modifiers": [], "Keys": ["KEY_F20"]}],
            "F21": [{"Modifiers": [], "Keys": ["KEY_F21"]}],
            "F22": [{"Modifiers": [], "Keys": ["KEY_F22"]}],
            "F23": [{"Modifiers": [], "Keys": ["KEY_F23"]}],
            "F24": [{"Modifiers": [], "Keys": ["KEY_F24"]}]
    }
}

Azkrath, 2020-09-03:
I have a portuguese keymap input working. To be able to use it you need to drop it at /usr/local/P4wnP1/keymaps folder.

mad4tv, 2021-01-24:
> I have a portuguese keymap input working. To be able to use it you need to drop it at /usr/local/P4wnP1/keymaps folder.

Hello! Can you please share your Portuguese keymap?
Thanks

Azkrath, 2021-01-24:
Hey man, sure.

Here is the one i am using:

`{
	"Name": "pt",
	"Description": "PT-pt (Portuguese) Keyboard Layout",
	"Mapping": {
		"CTRL": [{ "Modifiers": ["MOD_LEFT_CONTROL"], "Keys": [] }],
		"CONTROL": [{ "Modifiers": ["MOD_LEFT_CONTROL"], "Keys": [] }],
		"LEFT_CTRL": [{ "Modifiers": ["MOD_LEFT_CONTROL"], "Keys": [] }],
		"RIGHT_CTRL": [{ "Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": [] }],
		"ALT": [{ "Modifiers": ["MOD_LEFT_ALT"], "Keys": [] }],
		"LEFT_ALT": [{ "Modifiers": ["MOD_LEFT_ALT"], "Keys": [] }],
		"RIGHT_ALT": [{ "Modifiers": ["MOD_RIGHT_ALT"], "Keys": [] }],
		"SHIFT": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": [] }],
		"LEFT_SHIFT": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": [] }],
		"RIGHT_SHIFT": [{ "Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": [] }],
		"GUI": [{ "Modifiers": ["MOD_LEFT_GUI"], "Keys": [] }],
		"WIN": [{ "Modifiers": ["MOD_LEFT_GUI"], "Keys": [] }],
		"LEFT_GUI": [{ "Modifiers": ["MOD_LEFT_GUI"], "Keys": [] }],
		"RIGHT_GUI": [{ "Modifiers": ["MOD_RIGHT_GUI"], "Keys": [] }],

		"ESC": [{ "Modifiers": [], "Keys": ["KEY_ESC"] }],
		"ESCAPE": [{ "Modifiers": [], "Keys": ["KEY_ESC"] }],

		"F1": [{ "Modifiers": [], "Keys": ["KEY_F1"] }],
		"F2": [{ "Modifiers": [], "Keys": ["KEY_F2"] }],
		"F3": [{ "Modifiers": [], "Keys": ["KEY_F3"] }],
		"F4": [{ "Modifiers": [], "Keys": ["KEY_F4"] }],
		"F5": [{ "Modifiers": [], "Keys": ["KEY_F5"] }],
		"F6": [{ "Modifiers": [], "Keys": ["KEY_F6"] }],
		"F7": [{ "Modifiers": [], "Keys": ["KEY_F7"] }],
		"F8": [{ "Modifiers": [], "Keys": ["KEY_F8"] }],
		"F9": [{ "Modifiers": [], "Keys": ["KEY_F9"] }],
		"F10": [{ "Modifiers": [], "Keys": ["KEY_F10"] }],
		"F11": [{ "Modifiers": [], "Keys": ["KEY_F11"] }],
		"F12": [{ "Modifiers": [], "Keys": ["KEY_F12"] }],

		"SYSRQ": [{ "Modifiers": [], "Keys": ["KEY_SYSRQ"] }],
		"PRINT": [{ "Modifiers": [], "Keys": ["KEY_SYSRQ"] }],
		"PRINTSCR": [{ "Modifiers": [], "Keys": ["KEY_SYSRQ"] }],
		"SCROLLLOCK": [{ "Modifiers": [], "Keys": ["KEY_SCROLLLOCK"] }],
		"SCROLL": [{ "Modifiers": [], "Keys": ["KEY_SCROLLLOCK"] }],
		"PAUSE": [{ "Modifiers": [], "Keys": ["KEY_PAUSE"] }],
		"BREAK": [{ "Modifiers": [], "Keys": ["KEY_PAUSE"] }],
		"BRK": [{ "Modifiers": [], "Keys": ["KEY_PAUSE"] }],

		"\\": [{ "Modifiers": [], "Keys": ["KEY_GRAVE"] }],
		"1": [{ "Modifiers": [], "Keys": ["KEY_1"] }],
		"2": [{ "Modifiers": [], "Keys": ["KEY_2"] }],
		"3": [{ "Modifiers": [], "Keys": ["KEY_3"] }],
		"4": [{ "Modifiers": [], "Keys": ["KEY_4"] }],
		"5": [{ "Modifiers": [], "Keys": ["KEY_5"] }],
		"6": [{ "Modifiers": [], "Keys": ["KEY_6"] }],
		"7": [{ "Modifiers": [], "Keys": ["KEY_7"] }],
		"8": [{ "Modifiers": [], "Keys": ["KEY_8"] }],
		"9": [{ "Modifiers": [], "Keys": ["KEY_9"] }],
		"0": [{ "Modifiers": [], "Keys": ["KEY_0"] }],
		"'": [{ "Modifiers": [], "Keys": ["KEY_MINUS"] }],
		"«": [{ "Modifiers": [], "Keys": ["KEY_EQUAL"] }],

		"|": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"] }],
		"!": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_1"] }],
		"\"": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"] }],
		"#": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_3"] }],
		"$": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_4"] }],
		"%": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_5"] }],
		"&": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_6"] }],
		"/": [{ "Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_7"] }],
[... 135 lines cut ...]
		"RIGHT_ARROW": [{ "Modifiers": [], "Keys": ["KEY_RIGHT"] }],
		"LEFT_ARROW": [{ "Modifiers": [], "Keys": ["KEY_LEFT"] }],
		"DOWN_ARROW": [{ "Modifiers": [], "Keys": ["KEY_DOWN"] }],
		"UP_ARROW": [{ "Modifiers": [], "Keys": ["KEY_UP"] }],

		"COMPOSE": [{ "Modifiers": [], "Keys": ["KEY_COMPOSE"] }],

		"NUMLOCK": [{ "Modifiers": [], "Keys": ["KEY_NUMLOCK"] }],
		"NUM": [{ "Modifiers": [], "Keys": ["KEY_NUMLOCK"] }],

		"F13": [{ "Modifiers": [], "Keys": ["KEY_F13"] }],
		"F14": [{ "Modifiers": [], "Keys": ["KEY_F14"] }],
		"F15": [{ "Modifiers": [], "Keys": ["KEY_F15"] }],
		"F16": [{ "Modifiers": [], "Keys": ["KEY_F16"] }],
		"F17": [{ "Modifiers": [], "Keys": ["KEY_F17"] }],
		"F18": [{ "Modifiers": [], "Keys": ["KEY_F18"] }],
		"F19": [{ "Modifiers": [], "Keys": ["KEY_F19"] }],
		"F20": [{ "Modifiers": [], "Keys": ["KEY_F20"] }],
		"F21": [{ "Modifiers": [], "Keys": ["KEY_F21"] }],
		"F22": [{ "Modifiers": [], "Keys": ["KEY_F22"] }],
		"F23": [{ "Modifiers": [], "Keys": ["KEY_F23"] }],
		"F24": [{ "Modifiers": [], "Keys": ["KEY_F24"] }]
	}
}
`

mad4tv, 2021-01-24:
Thanks! @Azkrath