RoganDawes/P4wnP1_aloa #193: Problem connect to internet raspberry pi zero in windows 7 64 bits with putty

anduril-glitch, 2020-01-13:
Hello, 

I tried to connect to the internet my raspberry pi zero with P4wnPI ALOA under windows 7 but when I try to connect in ssh after having put the following ip address 172.16.0.1 ,

But putty tells me "Network Refused".

I don't understand and above all I can't find the "network-only.txt" payload ?

Thank you 

A +

anduril-glitch, 2020-01-13:
Please, I can't find an answer

mame82, 2020-02-06:
`172.16.0.1` is reachable if the RNDIS network adapter is working (USB only). So please check your network settings (specifically the configuration of the aforementioned RNDIS adapter).

For a WiFi connection to P4wnP1 ALOA's access point, the default IP is `172.24.0.1`.

`network-only.txt`doesn't apply to P4wnP1 ALOA, this is part of legacy P4wnP1

closed 2020-02-06 by mame82