RoganDawes/P4wnP1_aloa #197: Genimg

PineMaster, 2020-02-07:
What is the largest size that i can get drive created from genimg, also are the product and vendor id important or changable

flyer-machine, 2020-02-07:
Hi PineMaster,
I was also wondering about that, but it is possible to create a drive with 100 GB (and maybe more)
For me it was also important to store single big files, so I created a drive with NTFS. If you want an exfat file system, reformat the mounted container with a windows machine later.
A few modification in the original genimg did create a big ntfs container for me. You find the modified genimg enclosed.
It can be called like this: ./genimg_ntfs -i -s 100000 -o 100GB-NTFS -l alfa-store
(rename genimg_ntfs.txt to the same name without  the extension '.txt' before execution)

During the creation I always lost the network connection, but it finished the creation anyway after a while.

[genimg_ntfs.txt](https://github.com/mame82/P4wnP1_aloa/files/4170731/genimg_ntfs.txt)

Hope this helps.

Regards, Thomas

PineMaster, 2020-02-08:
Hey flyer, yeah i also lost internet connection tryna make a 8gb disk, it did make but not in FAt32 :[

closed 2020-02-09 by PineMaster