RoganDawes/P4wnP1_aloa #199: Es keymap not working properly

AnxoMJ, 2020-02-08:
When I try type | or other special chars they don´t appear (my pc is ussing Spanish from Spain keyboard)
I fixed already the "wrong chars", and this is the file "fixed" in case someone need it

`
{
	"Name": "ES",
	"Description": "Contains Spanish key mappings",
	"Mapping": {

            "CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "CONTROL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "LEFT_CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "RIGHT_CTRL": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": []}],
            "ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "LEFT_ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "RIGHT_ALT": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": []}],
            "SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "LEFT_SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "RIGHT_SHIFT": [{"Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": []}],
            "GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "WIN": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "LEFT_GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "RIGHT_GUI": [{"Modifiers": ["MOD_RIGHT_GUI"], "Keys": []}],

            "ESC": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],
            "ESCAPE": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],

            "F1": [{"Modifiers": [], "Keys": ["KEY_F1"]}],
            "F2": [{"Modifiers": [], "Keys": ["KEY_F2"]}],
            "F3": [{"Modifiers": [], "Keys": ["KEY_F3"]}],
            "F4": [{"Modifiers": [], "Keys": ["KEY_F4"]}],
            "F5": [{"Modifiers": [], "Keys": ["KEY_F5"]}],
            "F6": [{"Modifiers": [], "Keys": ["KEY_F6"]}],
            "F7": [{"Modifiers": [], "Keys": ["KEY_F7"]}],
            "F8": [{"Modifiers": [], "Keys": ["KEY_F8"]}],
            "F9": [{"Modifiers": [], "Keys": ["KEY_F9"]}],
            "F10": [{"Modifiers": [], "Keys": ["KEY_F10"]}],
            "F11": [{"Modifiers": [], "Keys": ["KEY_F11"]}],
            "F12": [{"Modifiers": [], "Keys": ["KEY_F12"]}],

            "SYSRQ": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINT": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINTSCR": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "SCROLLLOCK": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "SCROLL": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "PAUSE": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BREAK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BRK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],

            "º": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
            "1": [{"Modifiers": [],	"Keys": ["KEY_1"]}],
            "2": [{"Modifiers": [],	"Keys": ["KEY_2"]}],
            "3": [{"Modifiers": [],	"Keys": ["KEY_3"]}],
            "4": [{"Modifiers": [],	"Keys": ["KEY_4"]}],
            "5": [{"Modifiers": [],	"Keys": ["KEY_5"]}],
            "6": [{"Modifiers": [],	"Keys": ["KEY_6"]}],
            "7": [{"Modifiers": [],	"Keys": ["KEY_7"]}],
            "8": [{"Modifiers": [],	"Keys": ["KEY_8"]}],
            "9": [{"Modifiers": [],	"Keys": ["KEY_9"]}],
            "0": [{"Modifiers": [],	"Keys": ["KEY_0"]}],
            "'": [{"Modifiers": [],	"Keys": ["KEY_MINUS"]}],
            "¡": [{"Modifiers": [], "Keys": ["KEY_EQUAL"]}],
			
			"´": [{"Modifiers": [],	"Keys": ["KEY_APOSTROPHE"]},{"Modifiers": [], "Keys": ["KEY_SPACE"]}],
			"¨": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_APOSTROPHE"]}, {"Modifiers": [], "Keys": ["KEY_SPACE"]}],

            "ª": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
            "!": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_1"]}],
            "\"": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"]}],
            "·": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_3"]}],
[... 136 lines cut ...]
            "LEFT_ARROW": [{"Modifiers": [], "Keys": ["KEY_LEFT"]}],
            "DOWN_ARROW": [{"Modifiers": [], "Keys": ["KEY_DOWN"]}],
            "UP_ARROW": [{"Modifiers": [], "Keys": ["KEY_UP"]}],

            "COMPOSE": [{"Modifiers": [], "Keys": ["KEY_COMPOSE"]}],

            "NUMLOCK": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],
            "NUM": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],

            "F13": [{"Modifiers": [], "Keys": ["KEY_F13"]}],
            "F14": [{"Modifiers": [], "Keys": ["KEY_F14"]}],
            "F15": [{"Modifiers": [], "Keys": ["KEY_F15"]}],
            "F16": [{"Modifiers": [], "Keys": ["KEY_F16"]}],
            "F17": [{"Modifiers": [], "Keys": ["KEY_F17"]}],
            "F18": [{"Modifiers": [], "Keys": ["KEY_F18"]}],
            "F19": [{"Modifiers": [], "Keys": ["KEY_F19"]}],
            "F20": [{"Modifiers": [], "Keys": ["KEY_F20"]}],
            "F21": [{"Modifiers": [], "Keys": ["KEY_F21"]}],
            "F22": [{"Modifiers": [], "Keys": ["KEY_F22"]}],
            "F23": [{"Modifiers": [], "Keys": ["KEY_F23"]}],
            "F24": [{"Modifiers": [], "Keys": ["KEY_F24"]}]
    }
}

`

samu-delucas, 2020-12-09:
Do you simply replace the es.json with yours? And also, do you have to replace the one included in `~/P4wnP1/dist/keymaps` and the one in `/usr/local/P4wnP1/keymaps` or replacing only one of them is enough? Thanks!

HectorHumanes, 2022-08-13:
Many thanks @AnxoMJ it works like a charm! Would you mind if I put it on a MR to be included (hopefully one day) in the next release?. I'll credit you of course 🥇

AnxoMJ, 2022-08-13:
@HectorHumanes 
Sure, I'm glad it helped you.