RoganDawes/P4wnP1_aloa #204: Avoid writing to disk

Havunen, 2020-02-09:
Hey!

If I understood the code correctly all HID scripts save the input script to disk and then read it from there? I Would like to skip this step so I can mount the filesystem completitly read only. This way I could plug in and plug off the raspberry pi without worrying file system getting corrupted.

PineMaster, 2020-02-09:
First of all your scripts aer saved on a folder in sd card and that is NOT viewable by the computer unless you shove your sd card without a pi into a reader and plug in). SO there are absouletly no chances that you plug ur pi in and then it gets corrupted. n fact there is no disk emulated untill you enable the mass storage.
You see the computer recogonizes it as a keyboard NOT a storage device so it wont even see there is a micro sd card plugged in the pi, what the mass storage option does is that it emulates a harddrvive that is seen by the computer as a seperate device (pretty clever way of doing stuff, props to Mame82) so yeah no file system will by corrupted the comp cannot see ur scripts, they are stored in the SD card, the pc wont even know that you are reading of scripts because the sd card is not viewable to him !

Havunen, 2020-02-09:
My card has corrupted twice already so its real issue. I believe the corruption happens because I'm sending a lot HID scripts to rasp and then I would like to unplug the power at any time. 

Sources
https://mtyka.github.io/hardware/2018/10/09/read-only-pi.html
https://raspberrypi.stackexchange.com/questions/7978/how-can-i-prevent-my-pis-sd-card-from-getting-corrupted-so-often

PineMaster, 2020-02-10:
if you make the filesystem readable only you might end up destroying some logs files etc

Havunen, 2020-02-10:
Yes, I understand that. I consider it being less a problem than re-installing every week :/ ...

mame82, 2020-02-10:
Indeed script which are ran interactively from WebUI are transferred and stored as temporary file. Script execution, on the other hand, involves no write operations (neither for temporary scripts, nor for stored once).

System wide file IO operations, on the other hand, are buffered by kernel. Persisting pending file IO is usually done with the `sync` command. So having a Pi0 powered from an USB port only generally involves issues with data corruption.

Automated file system repair on boot has been disabled, to keep boot times short.

There are several options to mitigate data corruption:

- use `sync` when required
- shut down Pi 0 remotely to persist file IO ops
- run the Pi0 with additional external power (e.g. battery). In contrast to legacy P4wnP1 much effort has been put into keeping everything reconfigurable without power cycling the Pi. This even includes USB kernel patches, to allow re-triggering scripts when P4wnP1 is attached/detached to/from USB hosts, without power cycling it.

Note: For the DB a backend has been chosen, which syncs changes rapidly (to avoid corruption on power loss). Anyways, even the DB is file based in the end (just syncing more often)

Havunen, 2020-02-12:
Again my PI got corrupted... I did run shutdown command from SSH terminal before removing the usb cable but maybe it did not work because LED was on still...

21weberer, 2020-02-12:
I have a quick question regarding this. Is there a way to use the multiple python scripts that exist on the web to create the safe shutdown button, or would that interfere with the GPIO client, or not work at all?

21weberer, 2020-02-12:
@Havunen, you should most definitely wait until the light is off. Also, if you use linux as your main os, what I do is every time the Pi0 shuts down improperly, accidentally gets unplugged, etc. I put the sdcard into a reader, launch GParted, and check both partitions.