RoganDawes/P4wnP1_aloa #206: HID script stuck at running

mmm7761, 2020-02-12:
When I tried to run a HID script it's stuck at running and never runs it

Plz help as i tried it on both of my raspberry pie zero w and tried re flashing p4wnp1 over 5 times and it still didn't work

PineMaster, 2020-02-12:
First of all the usb has to be connected to a data port, and what script are you trying to use, try hidtest1.js first and see if it works, snake needs a bit of a rework for me(i still dont understand why lol) I cant think of anything else than u using the 1 port not the upper one (in between the hdmi and usb, use that)

21weberer, 2020-02-12:
Don't forget to use a data capable micro usb cable. I have done that by mistake before.

mmm7761, 2020-02-13:
My micro USB is data capable as I've transferred files from my phone to PC before
And I'm using the data port on my raspberry pie
And it's not working

21weberer, 2020-02-13:
Have you tried re-flashing the image?

mmm7761, 2020-02-14:
Yup Over 5 times

f0unix, 2020-02-18:
I came across the same issue, ensured it's a data cable, the PC recognizes the Pi0 but HID scripts are endless jobs and never run ... I am having doubts that this might be due to the MicroSD card used to flash on the image, I am using a 64GB Samsung Evo Plus !!

1byeworld, 2020-06-14:
same problem here still couldnt figure out a solution and this seems to happen on the earlier version of P4wnP1 too. 
and there is a another issue that has been closed on the same topic (https://github.com/RoganDawes/P4wnP1_aloa/issues/160)

this the link to the older version issue
https://github.com/RoganDawes/P4wnP1/issues/110

thecoder-001, 2020-06-14:
Unfortunately there is a small chance of your pi to be defective or have become damaged through the course of time. Mine's micro USB port got wore off due to repeated disconnecting and reconnecting. If that's the case then your pi would connect for a second, get disconnected again and the cycle would continue (however the power rail of the cable somehow keeps connected to the pi in my case and thus it remains online). 
The only solution of this problem is to either keep the pi very stable (don't move) while connected or solder a new micro-usb port instead of the old one (I think its possible, even if it might be hard)
PS: Before anyone tells me that I might not be using the data port, I can confirm that I am using the inner one. Also, the windows device management system is very very weird. It sometimes works and detects that the device is constantly getting disconnected and throws a warning or simply decides to remain quiet.

mmm7761, 2020-06-14:
> Unfortunately there is a small chance of your pi to be defective or have become damaged through the course of time. Mine's micro USB port got wore off due to repeated disconnecting and reconnecting. [...]

No I have recently purchased a new one and it still doesn't work

thecoder-001, 2020-06-14:
Oh, then it looks like that there is surely a software problem going on. I would try to replicate the issue on my pi and report back.
PS: Can you try to disable all the antivirus programs installed on your pc and try again? Another person had his P4wnP1 aloa behaving weirdly due to the antivirus suite on his pc (which probably somehow identified it as a rubber ducky and didn't allow it to function).

1byeworld, 2020-06-14:
ive tried it on multiple pcs and still dosent work and i know there is no issue with the pi because it connects to the pc and gives the Ethernet connection. but i haven't tried this without disabling the antivirus software, ill try that and comment later

this guy seems to have found a solution for the same problem occurring in the older version of P4wnP1 but i cant find the exact file on the P4wnP1 a.l.o.a

here is that comment if anyone figures out the lines in the new version pls let us know 
https://github.com/RoganDawes/P4wnP1/issues/110#issuecomment-331807315

but mame82 the creator himself replied to this comment and said this
https://github.com/RoganDawes/P4wnP1/issues/110#issuecomment-331819818

1byeworld, 2020-06-14:
OK i've been doing some research and i think that the version of the pi zero w might effect the actual program. Ive been using the Raspberry Pi Zero W V1.1 and if you guys are using the version i think that might be the thing in common we all have, if not please comment your version down below.

1byeworld, 2020-06-15:
Sorry if I'm focusing on this issue too much but i really want this to work and i've invested too much time and data to leave this now.😂

mmm7761, 2020-06-15:
> OK i've been doing some research and i think that the version of the pi zero w might effect the actual program. Ive been using the Raspberry Pi Zero W V1.1 and if you guys are using the version i think that might be the thing in common we all have, if not please comment your version down below.
>  
> Edit: Ok im not sure if there are any other versions other than the V1.1

I also have that version

1byeworld, 2020-06-15:
Ok I think that might be the problem but I'm not sure if that is the only version or if there is any other versions of the rpi zero w. I've been trying to figure this out for like 10 days now. I'm going to get another rpi today or tomorrow to check if my pi is the problem or if it is the image

thecoder-001, 2020-06-15:
> OK i've been doing some research and i think that the version of the pi zero w might effect the actual program. Ive been using the Raspberry Pi Zero W V1.1 and if you guys are using the version i [...]

I too have hardware revision version 1.1 and the P4wnP1 aloa software is working for me.
The rpi website (https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) lists only one revision of pi 0 w, but I surely remember that they did change something because when I ordered it, I head read few articles on that matter and was hoping to get the newer board layout

1byeworld, 2020-06-15:
Attach a picture of your board front and back I'll upload mine too then we can see what's the difference

thecoder-001, 2020-06-15:
> ive tried it on multiple pcs and still dosent work and i know there is no issue with the pi because it connects to the pc and gives the Ethernet connection. but i haven't tried this without disabling the antivirus software, ill try that and comment later
> 
> this guy seems to have found a solution for the same problem occurring in the older version of P4wnP1 but i cant find the exact file on the P4wnP1 a.l.o.a
> 
> here is that comment if anyone figures out the lines in the new version pls let us know
> [RoganDawes/P4wnP1#110 (comment)](https://github.com/RoganDawes/P4wnP1/issues/110#issuecomment-331807315)
> 
> but mame82 the creator himself replied to this comment and said this
> [RoganDawes/P4wnP1#110 (comment)](https://github.com/RoganDawes/P4wnP1/issues/110#issuecomment-331819818)

ok, it looks like I have found the code responsible for the fire onKeyboardUp. It is here: https://github.com/RoganDawes/P4wnP1_aloa/blob/master/hid/keyboard_led.go .
But just like Mame82, I think disabling the code won't do anything as it doesn't get called out (unless specifically asked to do so).

1byeworld, 2020-06-15:
Yeah but just try it and comment I'm not at home rn to try it and I'll also try asap

thecoder-001, 2020-06-15:
> Attach a picture of your board front and back I'll upload mine too then we can see what's the difference
Here is mine:
[image]
[image]

thecoder-001, 2020-06-15:
> Yeah but just try it and comment I'm not at home rn to try it and I'll also try asap

But, my P4wnP1_aloa pi is already working (I even reflashed my card 3-4 times to reproduce the bug but wasn't able to do so). [I am not saying that the bug doesn't exist or can't be reproduced, but I am simply not able achieve so]

Can both of you send screenshots of the web console while a script is running (take multiple screenshots if necessary)? I would like to see that, maybe we can get a clue.

1byeworld, 2020-06-15:
Ok my board is the same (I think I've attached pics) and if it's no trouble for you can you please upload the image file you have to a drive folder and share it.
I'm also attaching pics of every setting and  what the script says when I run it

1byeworld, 2020-06-15:
[image]
[image]

1byeworld, 2020-06-15:
[image]
[image]
[image]
[image]
[image]

thecoder-001, 2020-06-15:
In your board, there are some different logos. But other than that, I wasn't able to find any difference in the PCB layout.

1byeworld, 2020-06-15:
@thecoder-001 if you cant upload the image you have, please send the link of the file. all of the tutorials i watched they download the image from offensive - security but they have removed that image file from their website. If you can please try to upload the image file.

1byeworld, 2020-06-15:
> In your board, there are some different logos. But other than that, I wasn't able to find any difference in the PCB layout.

hmm but what if that's the problem.

thecoder-001, 2020-06-15:
1byeworld, try this: remove "waitLEDRepeat(NUM)" and run the script.
Like this:
`layout('us');			// US keyboard layout
typingSpeed(100,150)	// Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)

press("GUI r");
delay(500);
type("notepad\n")
delay(1000);
for (var i = 0; i < 3; i++) {
  type("Hello from P4wnP1 run " + i + " !\n");
  type("Moving mouse right ...");
  moveStepped(500,0);
  type("and left\n");
  moveStepped(-500,0);
}
type("Let's type fast !!!!!!!!!!!!!!!\n")
typingSpeed(0,0);
for (var i = 3; i < 10; i++) {
  type("Hello from P4wnP1 run " + i + " !\n");
  type("Moving mouse right ...");
  moveStepped(500,0);
  type("and left\n");
  moveStepped(-500,0);
}`

1byeworld, 2020-06-15:
you mean just run this

-------------------------------------------------------------------------------------------------------
press("GUI r");
delay(500);
type("notepad\n")
delay(1000);
for (var i = 0; i < 3; i++) {
type("Hello from P4wnP1 run " + i + " !\n");
type("Moving mouse right ...");
moveStepped(500,0);
type("and left\n");
moveStepped(-500,0);
}
type("Let's type fast !!!!!!!!!!!!!!!\n")
typingSpeed(0,0);
for (var i = 3; i < 10; i++) {
type("Hello from P4wnP1 run " + i + " !\n");
type("Moving mouse right ...");
moveStepped(500,0);
type("and left\n");
moveStepped(-500,0);
}`
-------------------------------------------------------------------------------------------------------

thecoder-001, 2020-06-15:
> you mean just run this
>
> ## press("GUI r");
> delay(500);
> type("notepad\n")
> delay(1000);
> for (var i = 0; i < 3; i++) {
> type("Hello from P4wnP1 run " + i + " !\n");
> type("Moving mouse right ..."); [...]

yes

thecoder-001, 2020-06-15:
which image file do you want? The os .iso?
Its available on the releases page: https://github.com/RoganDawes/P4wnP1_aloa/releases
here's the direct link: https://github.com/RoganDawes/P4wnP1_aloa/releases/download/v0.1.1-beta/kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa.img.xz

thecoder-001, 2020-06-15:
@1byeworld did it work?

1byeworld, 2020-06-15:
this is what i flashed 
im unable to try it right now since im out of my house when i go back ill try it and send you a message you 

and luckily i found a seller in my country with your type of board and since my country is in lock down it should be delivered to me by today evening or tomorrow evening

1byeworld, 2020-06-15:
and i tried all the software ( 2 ) in this page https://github.com/RoganDawes/P4wnP1_aloa/releases
and both of them dont work

thecoder-001, 2020-06-15:
well, the image files can't be corrupt as many use the same file. What is the OS of the test system?

thecoder-001, 2020-06-15:
> this is what i flashed
> im unable to try it right now since im out of my house when i go back ill try it and send you a message you
>
> and luckily i found a seller in my country with your type of board [...]

that's nice

1byeworld, 2020-06-15:
> 
> 
> well, the image files can't be corrupt as many use the same file. What is the OS of the test system?

windows 10 and ive tried it on windows 7 too and tried with and without virus guards

1byeworld, 2020-06-15:
> 
> 
> well, the image files can't be corrupt as many use the same file. What is the OS of the test system?

yeah the image file is not corrupt but i think i wont support some boards, maybe the boards were made for some countries

thecoder-001, 2020-06-15:
> > well, the image files can't be corrupt as many use the same file. What is the OS of the test system?
> 
> yeah the image file is not corrupt but i think i wont support some boards, maybe the boards were made for some countries

Can be. I was searching through rpi zero w on internet and I found 3 different board logo layouts. This is really confusing, they should have at least changed the hardware version or provided a easy guide to look for board changes.

1byeworld, 2020-06-15:
yes i also saw like 3 different boards its really weird, i know, but this seems to be the problem but i cant confirm until i get my hands on the other version of the pi, hopefully i get it by today and will be able to confirm the issue but until then i cant say anything.

hey @thecoder-001 when you ran for the first time like, when you booted the pi up for the first time with the P4wnP1 a.l.o.a image was the pi connected the internet or at some point before trying out and hid command, did you update && upgrade KALI. 
Because in one issue i saw someone say something about python on the image is not the newest version and that might be causing the problem.

1byeworld, 2020-06-15:
hmm but even if the processor is the same, cant the processor have different region issues.
sorry if my questions are really stupid, im new to this.

1byeworld, 2020-06-15:
> 
> 
> hmm but even if the processor is the same, cant the processor have different region issues.
> sorry if my questions are really stupid, im new to this.

like the same processor could be manufactured for different regions specifically

1byeworld, 2020-06-15:
@thecoder-001 dude and if you followed a videos when you installed pls attach the link and if you didn't and if it is not too much of a problem can you explain how you did it

00Pille, 2020-06-15:
Hi,
just tried to rebuild the Error. Can you disable CDC ECM than check Device Manager for any unknown Devices ? Thats what caused my Error.

1byeworld, 2020-06-15:
@thecoder-001 @00Pille ok thecoder you were right. it was working all this time i just had to delete this line "waitLEDRepeat(NUM)" it worked on the old pi too and i'm relieved and thank you for all the help. OMFG im just soo happy i cant thank you enough.

1byeworld, 2020-06-15:
[image]
thanx again soo much @thecoder-001

thecoder-001, 2020-06-15:
No problem, I am happy that I was able to help you.

Lemme explain the doubts you had (just because, what's the benefit of something working if you don't know what & how it got fixed?) :
1)
> hmm but even if the processor is the same, cant the processor have different region issues.
> sorry if my questions are really stupid, im new to this.

The rpi organization sources all the processors from Broadcom having the same model no. There can be regional variations but they all exist in software & **not in hardware** (eg- different countries have different wifi channels, which can be changed on pi through config). So, programs should have no problems on running on different boards. [Btw, its okay to ask stupid questions, but its stupid to not ask questions]

2)**Why did it work:**
See this: https://github.com/RoganDawes/P4wnP1_aloa#wait-for-led-report
After reading the above, you might be thinking that why it is not working then? Well it is not supported on macOS and windows...because windows is windows. Windows has over time excelled in the field of showing unpredictable behavior with external devices and thus behaves strangely. Sometimes its not able to install the driver, sometimes it doesn't even bother to do anything and sometimes it decides to remain quiet after initializing the device. The list continues. So for mission-critical operations (read: mission-critical tests on authorized machines), it is better to just add a delay to give time for windows to initialize the pi as keyboard & mouse.

Hope this helps.

1byeworld, 2020-06-16:
Thanks I'll look into them and once again thanx for all the help

DrClip, 2023-03-18:
unfortunately for me, none of the above explanations eliminated the problem I had regarding not being able to run any script, the error message was the following : 

**_Error starting script as background job
rpc error: code = Unknown desc = HIDScript not available (mouse and keyboard disabled)_**

after a few failed attempts I realized that the problem is neither the board nor any other associated problem, but the image that I flashed on the memory card, the image taken from https://kali.download/arm-images/kali-2023.1/kali-linux-2023.1-raspberry-pi-zero-w-p4wnp1-aloa-armel.img.xz does not work for me (kali-linux-2023.1-raspberry-pi-zero-w-p4wnp1-aloa-armel) ----- **(not good 👎 )**
the image that works for me is the one that comes from github (https://github.com/RoganDawes/P4wnP1_aloa   
kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa) ----- **(good one 👍)**

ZIKH26, 2023-10-12:
A few days ago, I encountered the issue where the HID script was continuously running and couldn't execute successfully. After two days of troubleshooting, I borrowed six USB data cables from my friend, and, in the end, one of them allowed the HID script to run successfully. Therefore, I can confirm that this issue is definitely related to the USB cable. Just having the ability to transfer data doesn't guarantee the success of the HID script execution, as all six cables were capable of data transmission. It might be due to some unknown reasons, but during my testing, it was indeed related to the data cable.
English is not my native language, so the text translated by a machine may appear a bit unusual.

ZIKH26, 2023-10-13:
@MenachemMarozov   Unfortunately, this cable is from a few years ago when my friend purchased a Huawei phone with the model HRY-AL00a. While I can now find cables online that are compatible with the HRY-AL00a model, they are not the original ones, and I can't guarantee that these cables will work. Therefore, I can't share a link with you. I think trying a few different USB cables might be successful.