RoganDawes/P4wnP1_aloa #207: Genimg question

Noidea666, 2020-02-12:
have anyone ever tryied making bootable usb in genimg partition? is it even possible mostly wanted to see pogstick and kon boot on P4WNP1 a.l.o.a. (maybe even windows 10 bootable usb)

21weberer, 2020-02-12:
Theoretically, it is possible. Here is what you should do:
make a drive with genimg (the *.bin file), 
mount the drive using the web/ssh client, 
then create a bootable drive from the host os. So if you have P4wnP1 and use windows, mount the virtual disk in windows, and configure it from there. The last part of your post is a little confusing, could you clarify?

Noidea666, 2020-02-14:
Thank you for the answer. I tested to install windows 10, kalilinux x86,  pogostick and konboot to genimg partition. everything but konboot is working fine. Konboot wont show up as bootable usb at startup. (no idea why)

21weberer, 2020-02-23:
Ok, ill do some research.