RoganDawes/P4wnP1_aloa #209: Windows 10 freezes

H4CK3RxDE, 2020-02-14:
Hello, when I plug my P4wnP1 in my PC, the PC freezes and I can't do anything until I disable Keyboard-HID in the USB settings of P4wnP1. I'm on Windows 10 Version 1909 Build 18363. Hope someone can help me. thanks in advance.

Havunen, 2020-02-15:
Hi @H4CK3RxDE I'm on same build of windows 10 and I'm not experiencing any issues. Try changing USB serial / manufacturer / identifier to see if its conflicting with some driver.

H4CK3RxDE, 2020-02-16:
I changed VendorID, ProductID, Manufacturer... , but this didn't help. Keeps freezing. the only thing that does not freeze is music played on the pc everything else like mouse, keyboard and animated background freezes.

H4CK3RxDE, 2020-02-16:
I dont know if its important but when I change something in the usb settings and deploy it it freezes and than theres an error message who says "error while deploying new USB gadget settings  rpc error: code = Unknown desc = Response closed without headers"

H4CK3RxDE, 2020-02-18:
Any further help?

ayush06feb, 2020-05-16:
It might be that your PC has some antivirus that identifies the pi as rubber ducky and tries to protect the computer but causes damage itself. Try disabling it and try again.
For the USB deploying error, I also get it sometimes but repeated trying and reboot gets me past it. Maybe try to reflash the latest image.

H4CK3RxDE, 2020-05-16:
@ayush06feb Thanks a lot, that worked. It was the antivirus program. Why didn't I think of that myself?

closed 2020-05-16 by H4CK3RxDE