RoganDawes/P4wnP1_aloa #210: Possibility to Install a lightweight GUI?

mukhtharcm, 2020-02-15:
Can anyone recommend an simple way to install a **_**GUI( desktop )**_** into this? Possibly with a vnc server so i can use as a real portable desktop.

PineMaster, 2020-02-16:
I dont think that would be possible because that would ditch(change) many settings that p4wnp1 uses to function and its probably of no use to carry a GUI on a pi zero w

SymbianSyMoh, 2020-02-16:
Technically you can, I did it and installed tightvncserver without breaking any of P4wnP1's functionality so i had a full Kali experience with desktop access and all P4wnP1 features but then removed it because i saw no necessity for it + it will eat a bit of your Pi's zero ram which is already enough to have a good performance without GUI!

But if you want to do it anyway, connect your P4wnP1 to internet (via Ethernet bridging, Ethernet Dongle or via Wifi) then `apt-get install tightvncserver` then `vncserver` and configure it.

mukhtharcm, 2020-02-17:
@SymbianSyMoh which leghtwight desktop you used it?
And can i just remove with just 
```apt remove tightvncserver``` ?

SymbianSyMoh, 2020-02-17:
@mukhtharcm I used XFCE, It's the lightest i guess. When you want to get rid of both just remove `tightvncserver` and `xfce` window manager!

mukhtharcm, 2020-02-17:
@SymbianSyMoh  but when i try to install xfce, it shows i have to remove console-common, Is there any problem doing this?

SymbianSyMoh, 2020-02-17:
@mukhtharcm Have never faced this, my advise is to backup by creating an .img of the whole SD card of P4wnP1's installation then you can play however you want with this topic.

mukhtharcm, 2020-02-17:
@SymbianSyMoh then how can i create .img of the whole sd card?

SymbianSyMoh, 2020-02-17:
Do me a favor and Google it, there are tons of tutorials about this topic!

mukhtharcm, 2020-02-18:
@SymbianSyMoh  thank you for your help and sorry for interupting.

closed 2020-02-18 by mukhtharcm