RoganDawes/P4wnP1_aloa #212: Testing new keyboard layout

TheMMcOfficial, 2020-02-16:
I have try to test my layout multiple time but they don't get loaded. I realise that the layout is not right when I run my test script: 

````
layout('cm');
typingSpeed(100,150)	// Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)

type("abcdefghijklmnopqrstuvwxyz ABCDEFGHIJKLMNOPQRSTUVWXYZ 1234567890 !@#$%?&*()-=_+");
```
`

I have also got the confirmation in syslog that the layout can't be load: 

````
Feb  6 19:18:19 kali P4wnP1_service[173]: START JOB 2 SCRIPT ON VM 0
Feb  6 19:18:19 kali P4wnP1_service[173]: 2020/02/06 19:18:19 HIDScript layout: Setting layout to 'cm'
Feb  6 19:18:19 kali P4wnP1_service[173]: 2020/02/06 19:18:19 HIDScript layout: Couldn't set layout `cm`: Language map with name 'cm' isn't loaded!
Feb  6 19:18:19 kali P4wnP1_service[173]: 2020/02/06 19:18:19 HIDScript type: Typing 'abcdefghijklmno ...' on HID keyboard device '/dev/hidg0'
Feb  6 19:18:33 kali P4wnP1_service[173]: JOB 2 on VM 0 SUCCEEDED WITH RESULT: null`
```

Noidea666, 2020-02-18:
can you paste here the layout so we can take a look at it or you can try to figure it out yourself. usually you have to do little chages to get it working.

TheMMcOfficial, 2020-02-18:
ca: https://github.com/RoganDawes/P4wnP1_aloa/blob/master/dist/keymaps/ca.json
cm: https://github.com/RoganDawes/P4wnP1_aloa/blob/master/dist/keymaps/cm.json

Noidea666, 2020-02-22:
i tested those but can't get the keymaps work. i even made ca.json again from fi.json keymap layout.
i had the same problem while i made fi.json and MacFI.json. fixed it by remaking the file from different .json file what i found from keymaps folder.

TheMMcOfficial, 2020-02-22:
Ok is it because it have some weird encoding? Do I need to rewrite everything or just copy paste should work ?

Noidea666, 2020-02-22:
im not sure but you could try using the common.json layout and modify it to canadian layout.

here's the layout:

{
	"Name": "COMMON",
	"Description": "Contains common key mappings which are the same in most languages",
	"Mapping": {

            "CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "CONTROL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "LEFT_CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "RIGHT_CTRL": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": []}],
            "ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "LEFT_ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "RIGHT_ALT": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": []}],
            "SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "LEFT_SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "RIGHT_SHIFT": [{"Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": []}],
            "GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "WIN": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "LEFT_GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "RIGHT_GUI": [{"Modifiers": ["MOD_RIGHT_GUI"], "Keys": []}],

            "ESC": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],
            "ESCAPE": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],

            "F1": [{"Modifiers": [], "Keys": ["KEY_F1"]}],
            "F2": [{"Modifiers": [], "Keys": ["KEY_F2"]}],
            "F3": [{"Modifiers": [], "Keys": ["KEY_F3"]}],
            "F4": [{"Modifiers": [], "Keys": ["KEY_F4"]}],
            "F5": [{"Modifiers": [], "Keys": ["KEY_F5"]}],
            "F6": [{"Modifiers": [], "Keys": ["KEY_F6"]}],
            "F7": [{"Modifiers": [], "Keys": ["KEY_F7"]}],
            "F8": [{"Modifiers": [], "Keys": ["KEY_F8"]}],
            "F9": [{"Modifiers": [], "Keys": ["KEY_F9"]}],
            "F10": [{"Modifiers": [], "Keys": ["KEY_F10"]}],
            "F11": [{"Modifiers": [], "Keys": ["KEY_F11"]}],
            "F12": [{"Modifiers": [], "Keys": ["KEY_F12"]}],

            "SYSRQ": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINT": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINTSCR": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "SCROLLLOCK": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "SCROLL": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "PAUSE": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BREAK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BRK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],

            "`": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
            "1": [{"Modifiers": [],	"Keys": ["KEY_1"]}],
            "2": [{"Modifiers": [],	"Keys": ["KEY_2"]}],
            "3": [{"Modifiers": [],	"Keys": ["KEY_3"]}],
            "4": [{"Modifiers": [],	"Keys": ["KEY_4"]}],
            "5": [{"Modifiers": [],	"Keys": ["KEY_5"]}],
            "6": [{"Modifiers": [],	"Keys": ["KEY_6"]}],
            "7": [{"Modifiers": [],	"Keys": ["KEY_7"]}],
            "8": [{"Modifiers": [],	"Keys": ["KEY_8"]}],
            "9": [{"Modifiers": [],	"Keys": ["KEY_9"]}],
            "0": [{"Modifiers": [],	"Keys": ["KEY_0"]}],
            "-": [{"Modifiers": [],	"Keys": ["KEY_MINUS"]}],
            "=": [{"Modifiers": [], "Keys": ["KEY_EQUAL"]}],

            "~": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
            "!": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_1"]}],
            "@": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"]}],
            "#": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_3"]}],
            "$": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_4"]}],
            "%": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_5"]}],
            "^": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_6"]}],
[... 108 lines cut ...]
            "UP": [{"Modifiers": [], "Keys": ["KEY_UP"]}],
            "RIGHT_ARROW": [{"Modifiers": [], "Keys": ["KEY_RIGHT"]}],
            "LEFT_ARROW": [{"Modifiers": [], "Keys": ["KEY_LEFT"]}],
            "DOWN_ARROW": [{"Modifiers": [], "Keys": ["KEY_DOWN"]}],
            "UP_ARROW": [{"Modifiers": [], "Keys": ["KEY_UP"]}],

            "COMPOSE": [{"Modifiers": [], "Keys": ["KEY_COMPOSE"]}],

            "NUMLOCK": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],
            "NUM": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],

            "F13": [{"Modifiers": [], "Keys": ["KEY_F13"]}],
            "F14": [{"Modifiers": [], "Keys": ["KEY_F14"]}],
            "F15": [{"Modifiers": [], "Keys": ["KEY_F15"]}],
            "F16": [{"Modifiers": [], "Keys": ["KEY_F16"]}],
            "F17": [{"Modifiers": [], "Keys": ["KEY_F17"]}],
            "F18": [{"Modifiers": [], "Keys": ["KEY_F18"]}],
            "F19": [{"Modifiers": [], "Keys": ["KEY_F19"]}],
            "F20": [{"Modifiers": [], "Keys": ["KEY_F20"]}],
            "F21": [{"Modifiers": [], "Keys": ["KEY_F21"]}],
            "F22": [{"Modifiers": [], "Keys": ["KEY_F22"]}],
            "F23": [{"Modifiers": [], "Keys": ["KEY_F23"]}],
            "F24": [{"Modifiers": [], "Keys": ["KEY_F24"]}]
    }
}

TheMMcOfficial, 2020-09-23:
Issue resolve here: https://github.com/RoganDawes/P4wnP1_aloa/issues/257

TheMMcOfficial, 2020-09-23:
[CA.zip](https://github.com/RoganDawes/P4wnP1_aloa/files/5268254/CA.zip)

closed 2020-09-23 by TheMMcOfficial