RoganDawes/P4wnP1_aloa #214: Installation

ChopsKingsland, 2020-02-16:
How do I install this, because it just tries to write a "test.bin" file

PineMaster, 2020-02-16:
You have to go to releases and then find the latest version and download the img, you have to use etcher to flash it onto a sd card (i use etcher) then chuck the sd card in the pi, a ssid will appear soo in wifi selection, its name should be well obvious its a bunch of emojis and the PSK is MaMe82-P4wnP1, go to 172.24.0.1:8000 and you can check there or u can just follow this vid https://www.youtube.com/watch?v=I_BjCdJlCo4 
;]

ChopsKingsland, 2020-02-16:
Is it the one that is .img.xz?

PineMaster, 2020-02-16:
> Is it the one that is .img.xz?

Yup!

brandonplank, 2020-06-21:
the newest version seems corrupted for me. I have downloaded it two time and i have gotten two different checksums 
Brandons-MBP:~ brandonplank$ shasum -a 256 /Users/brandonplank/Downloads/kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa.img.xz 
b4aa3b41261a8be503e98cbd99ceab8b8e8119eb3301f6fb21dc59d399355b5e  /Users/brandonplank/Downloads/kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa.img.xz

Brandons-MBP:~ brandonplank$ shasum -a 512 /Users/brandonplank/Downloads/kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa\ \(1\).img.xz 
3a3e3f6df0d234987fad102e391d828b2cb9daf1c4e7333f09e614c515a701ebc7cc8ed24fddf03bdb004bef12970d8c1cfd0bfba24aa7b2c2c499323407c6de  /Users/brandonplank/Downloads/kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa (1).img.xz

brandonplank, 2020-06-21:
I cannot find the checksums to these files online.