RoganDawes/P4wnP1_aloa #215: USB on PC

ChopsKingsland, 2020-02-17:
After a successful install, I plug it into my pc, and it doesn't even show up! how can i fix this?

PineMaster, 2020-02-17:
You have to look into the available wifis ......

ChopsKingsland, 2020-02-17:
ive connected to the wifi on my phone, and im trying to run an HID script. It says it ran successfully, but nothing happens, and it doesnt show up on my pc.

PineMaster, 2020-02-17:
did u use the data ports?
and check the script, if you are running Hidtest.1 then you have to place the cursor on ur pc somewhere

PineMaster, 2020-02-17:
try saving this script https://pastebin.com/zjst6Af3 and running it, its a very basic script that opens notepad and types that ur p4wnp1 is working :]
Happy pwn'ng!

ChopsKingsland, 2020-02-17:
it just says its running but nothing is happening, could it be a probelm with my pc?

ChopsKingsland, 2020-02-17:
it cant be my pc, because it doesnt work on my laptop either. what are the correct settings for hid?

PineMaster, 2020-02-17:
did you follow the vid i gave you in ur earlier issue, follow that first and test the script i gave you

ChopsKingsland, 2020-02-17:
i followed the video a few months ago and it still didnt work. i followed it again and it still doesnt work

PineMaster, 2020-02-17:
which script u testing?

ChopsKingsland, 2020-02-17:
the one you sent

ChopsKingsland, 2020-02-17:
i tried re-imaging the card and that hasnt sorted it either

PineMaster, 2020-02-17:
did you disable mouse and enable mass storage (create disk from genimg)?

ChopsKingsland, 2020-02-18:
yep i followed the whole tutorial and that didn't work ＞﹏＜

ChopsKingsland, 2020-02-18:
@RoganDawes any ideas how to get it working?

RoganDawes, 2020-02-18:
Let’s start at the beginning:
Disable all of the USB interfaces on the P4wnP1, then enable the keyboard and mouse, and SAVE that as the startup profile.
Reboot the Pi (or else apply the configuration), and check if you can see the keyboard/mouse on your target.

ChopsKingsland, 2020-02-18:
ive done that and it doesnt work

the 'job' just stays at RUNNING
[image]

RoganDawes, 2020-02-18:
Please can you verify that the cable is in fact capable of transferring data, and does not merely have the two power conductors of a "charge-only" cable? Depending on your OS, you could also check to see if the P4wnP1 HID devices are actually seen by the host. On Linux, you can run `dmesg -w` to see the messages as it is enumerated (run `dmesg -w` first, then plug the P4wnP1 in and wait for it to boot). On Windows you can check with the Device Manager to see what USB devices have been recognized, or you can use USBDeView (https://www.nirsoft.net/utils/usb_devices_view.html).

If nothing shows up, there are two possibilities: Either P4wnP1 is not properly creating the USB devices, or else the cable is the problem. Try with a different cable (or several different ones if you still have problems). Check with a phone or similar device if that is recognized by the Host OS when using that cable. Etc

ChopsKingsland, 2020-02-18:
i have tried multiple cables, and that has not helped. it doesnt show up in device manager either...

evilbobbins, 2020-02-19:
Are you plugging in the data usb to the pc or just the power one?

PineMaster, 2020-02-19:
Yeah seems like a hardware problem to me :p

ChopsKingsland, 2020-02-19:
ive tried using a new sd card with just raspian on, and the USB data port doesnt detect a keyboard or mouse. it still accepts power though...

RoganDawes, 2020-02-19:
Attach a picture of your setup (cables, etc) so we can get a better idea.

ChopsKingsland, 2020-02-19:
[image]
I just plug this into USB. I’ve also tried it on my laptop and it doesn’t work. I have also tried with numerous cables too

RoganDawes, 2020-02-19:
Ok. At least you are using the right port. I haven't played with the latest Raspbian, but as far as I recall, there is no default USB device configuration, so expecting to see one was misplaced.
If you have verified that the cable is working as a data cable with some other device, then I'm afraid I am out of ideas.

PineMaster, 2020-02-19:
Other tha that it could only be the pi thats not able to send over data if you followed all of these comments and the vid, sorry i couldnt help

Maow0, 2020-04-15:
I had the exact same issue tried a bunch of cables that didnt work, had to order a cable from amazon specifically with data lines. Problem solved! maybe you just got real unlucky with your cables check to see if when you plug in your phone etc with those same cables that they defo get picked up on your host OS's device manager.

closed 2020-04-25 by RoganDawes