RoganDawes/P4wnP1_aloa #216: Connecting p4wnp1 to the internet

PineMaster, 2020-02-19:
How can i connect p4wnp1 to the internet?

RoganDawes, 2020-02-19:
You have a couple of options for connecting to the Internet. 
1. You can configure your P4wnP1 to connect to an existing Wi-Fi AP with routing to the Internet. This is probably the simplest option,
2. You can connect P4wnP1 to your laptop using USB, and configure your laptop to route P4wnP1 traffic. This can get pretty complicated, depending on what OS you are using, especially since the P4wnP1 is typically configured to issue DHCP addresses itself, whereas "connection sharing" or OS-equivalent technology generally configures your laptop as the DHCP server.
3. You can add an additional interface to your P4wnP1, for example a 3G modem using the OTG port, and configure that as required.
4. Or you can port forward traffic using SOCKS over an SSH connection to the P4wnP1.

It ultimately depends on exactly you want to achieve.

w.r.t. the OLED menu, I personally have no experience with this, and would hope for someone else to jump in and assist with suggestions.

PineMaster, 2020-02-19:
> You have a couple of options for connecting to the Internet.
>
> 1. You can configure your P4wnP1 to connect to an existing Wi-Fi AP with routing to the Internet. This is probably the simplest option, [...]

I would like to do the 1st method, how can i configure my p4wnp1 o an existing Wi-Fi AP with routing to the Internet ?

RoganDawes, 2020-02-20:
https://p4wnp1.readthedocs.io/en/latest/P4wnP1-W/Wifi-Client/

closed 2020-02-20 by RoganDawes