RoganDawes/P4wnP1_aloa #217: HID Scripts Are Not Running

wowkster, 2020-02-22:
Every time I try to run an HID script it gives me a "CONTROLLER ABORTED" message. This strangely only really happens when the keyboard is used as I can run the "cosmouse" and "sinmouse" scripts without a problem. 

I have tried reflashing multiple times, even going back to previous release versions to no avail. I also have tried everything suggested in this thread: [https://github.com/RoganDawes/P4wnP1_aloa/issues/160](url) but I was not able to so easily fix the problem. I will provide a picture of the error.
[image]

PineMaster, 2020-02-22:
those are mouse, scripts make sure u have the mouse emulated

wowkster, 2020-02-22:
I do have them enabled. Everything is setup as per the instructions as well as some videos I found.

PineMaster, 2020-02-22:
> only really happens when the keyboard is used as I can run the "cosmouse" and "sinmouse" scripts
wdym by that, are u using the keyboard?
ive only experienced the aborted thing when i spam scripts, do you have like a bunch of scripts spammed at one time?

wowkster, 2020-02-24:
I’m not spamming scripts. It happens even if I only have one running

PineMaster, 2020-02-25:
Oh okk i was misunderstanding ur problem (sorry my English :p) ,  try disabling mouse and only put RNDIS and keyboard on usb settings or just post a photo of ur settings here