RoganDawes/P4wnP1_aloa #221: GB Keymap not correctly working on windows,

techtino, 2020-02-27:
Any keys which are accessed using MOD_ALT_RIGHT just don't work. Windows seems to block this functionality for some reason, however my Linux install works with it perfectly. Its strange, if I switch to US keyboard layout on windows and in my script, it works perfectly. Examples of these keys are: \,{,},[,]...
[image]

I'm not certain if this is entirely the cause, all I know is that under Linux, these keys are being entered correctly.

Thinking I can solve it by changing the keymap, but I don't know what the regular keys for it on the GB keyboard is.

techtino, 2020-02-27:
Solved this issue by reassigning the keys I cared about to the correct keys that work under windows. Used EVTEST on my linux machine to find the key name for the keymap. E.g. \ was changed to KEY_102ND. Created a new keymap and it worked. 
[gb-2.txt](https://github.com/RoganDawes/P4wnP1_aloa/files/4261796/gb-2.txt)

bbzrta, 2020-03-06:
thanks a lot techtino your keymap really helped me out.

ajc0502, 2020-03-13:
Where do we replace the keymap file? I'm replacing gb.json with your keymap (and renaming it to keymap.json) in "P4wnP1/dist/keymaps/" and I'm still getting the same error

bbzrta, 2020-03-13:
> 
> 
> Where do we replace the keymap file? I'm replacing gb.json with your keymap (and renaming it to keymap.json) in "P4wnP1/dist/keymaps/" and I'm still getting the same error

hey ajc0502, you have to put the json file in this directory "**/usr/local**/P4wnP1/keymaps". 
and for calling it you should say "layout('gb-2');"

(unless of course, you rename the file.)

ajc0502, 2020-03-13:
THANK YOU so much bbztraI was up all morning trying to figure it out

PsymonsB, 2020-11-28:
thanks for this was struggling to understand why some hid scripts weren't working on gb layout may look over all the key mappings to check them