RoganDawes/P4wnP1_aloa #223: Requirements of custom bash script for a trigger action?

ritchie-tg, 2020-03-06:
Hello all,
     I wanted to simply create a bash script that would change the led blinking rate when dhcp was connected. I just got started with the latest beta image and trying to familiarize myself with everything. I was having some issues where th Pi wouldn't connect to my wifi sometimes. A different blink rate would tell me which address it has - without having to check the dchp list of the router every time. 

     This seems pretty straight forward. I found the other existing bash scripts, created a similar file.sh in that directory as you normally do; shebang and chmod +x. Permissions are identical to the others. I reboot the Pi Zero W. But the dhcp-connect.sh I made is not in the list under triggeractions > action > run a bash script. Just:

serial-temrinal.sh
servicestart.sh
and so on....

      I imagine being new I'm overlooking something. But I cannot imagine making this process of adding a custom bash script to be much more complicated than this?

PineMaster, 2020-03-06:
did you save it to `/usr/local/P4wnp1/scripts` ?

ritchie-tg, 2020-03-06:
> did you save it to `/usr/local/P4wnp1/scripts` ?

No. I didn't even know that directory existed. I found the "./P4wnP1/dist/scripts/" and did not anticipate duplicates.

I moved my script to the directory you suggested and it now shows up in the web UI. This will ease the process of learning the framework for sure. Thank you!

[image]

closed 2020-03-06 by ritchie-tg