RoganDawes/P4wnP1_aloa #225: Wifi does not appear

crocogab, 2020-03-08:
Hello
I would already like to congratulate you for the great tool that you offer us.
Unfortunately I bought a rapsberry pi 0 w recently and after having flashed the bone and plugged in my raspberry no new wifi network appears.
I don't know where it can come from.
When I take the same sd card with the same system but I put it this time in my raspberry pi 3 I have no problem with wifi.
Thank you in advance.

PineMaster, 2020-03-10:
Are you sure its a Zero w?

crocogab, 2020-03-10:
I bought it on the official site of raspberry normally it is a raspberry pi 0 w

PineMaster, 2020-03-11:
try using an external wifi adapter, if it does than the pi's antenna is broken

crocogab, 2020-03-11:
Can I get a refund?
Thank you very much for your answer

V0r-T3x, 2020-08-24:
@crocogab 

I was some little problems with wifi too. Try to connect to the Pi0 with ssh and type `DHClient` or `dhcpcd`. It work for me.

crocogab, 2020-08-25:
> @crocogab
> 
> I was some little problems with wifi too. Try to connect to the Pi0 with ssh and type `DHClient` or `dhcpcd`. It work for me.

Ty will try thx