RoganDawes/P4wnP1_aloa #226: Possibility of Pwnagotchi?

PineMaster, 2020-03-10:
Has anyone been able to import p4wnagotchi to this, i've been trying it for 2 days and no success, is some p4wnp1 feature breaking pwnagotchi on it?

Thanks

Jan200101, 2020-03-14:
connecting over the usb data connection will stop pwnagotchi from doing its job

PineMaster, 2020-03-14:
ik and after more research i found out pwnagotchi requires "tensorflow" and that is not supported by kali's packages, pwnagotchi is based on raspbain so i dont think i will be able to get it workin :[

Casul51, 2020-03-15:
Anyway you can use old P4wnP1 as I am trying to use it alongside with retropie. Probably I will add some features to using wifite and autopwn nearby WPS and WEP wifies. (but I need to make script to auto pwn good wifies (like +60db) with connecting it somehow with my wifi)