RoganDawes/P4wnP1_aloa #227: too slow to type all keys.

Noidea666, 2020-03-23:
okay so this happened to me today my sdcard got corrupted when unplugged P4wnP1 from computer and i got the newer version of p4wnp a.l.o.a. 

now when i use 

layout('FI');		
typingSpeed(0,0)	
type("qwertyuiopåasdfghjklöäzxcvbnm QWERTYUIOPÅASDFGHJKLÖÄZXCVBNM , . - < > | ! \"  ¤ % & /  ) (  = ? + ´ ` ¨ ^ * § ½ @ £ $ € { [ ] \\ }") 

all the keys are not typed but whe i change to typingSpeed(10,0) it's working just fine.

can anyone tell why it's doing this? on older version when i made the fi.json and tested it worked just fine while using typingSpeed(0,0). 

Thank you.

PineMaster, 2020-03-24:
Probably becuz of the base 64 things use `' smtn '` instead of `" smth "` worked for me.

Noidea666, 2020-03-24:
i have tested that no difference.

PineMaster, 2020-03-26:
Hello, apparently p4wnp1 cant write everything cuz its not defined in the keymaps you can look at this https://github.com/RoganDawes/P4wnP1_aloa/blob/master/dist/keymaps/US.json file and change it it in `/usr/local/P4wnP1/keymaps/common.json` and be sure to select the US (or whatever lang u edited) in ur script @Noidea666

Noidea666, 2020-04-09:
the computer i tested with turned out to be the problem..

closed 2020-04-09 by Noidea666