RoganDawes/P4wnP1_aloa #233: Mass storage USB size has only ~4M space

holmesgithub, 2020-04-13:
Hi I used this OS on RPI Zero W successfully and it's great. Completed all install and setup, I plugged into my Windows 10 PC and it is recognized as a ramdrive with assigned drive letter "E:" which is wonderful. But it only has ~4M space. 
Since the document about the USB mass storage says "...Currently only in demo setup with 128 Megabyte drive". What can I do to increase the "E:" drive to have more space like 1G?
P.S. my microSD card on RPI is 32GB.

thank you very much!

PineMaster, 2020-04-14:
K dude, did you bother watching seytonics vid becuz he mentons it ther, neverthenless, the command:
` sudo /usr/local/P4wnP1/helper/genimg -i -s <size in mb> -o <smth like 5G or 500mb> -l <name of drive>`
More info- https://www.youtube.com/watch?v=I_BjCdJlCo4&t=430s
then u can change current drive from the p4wnp1 web menu
happy pwning!

holmesgithub, 2020-04-14:
Thank you so very much for the info PineMaster! definitely give it a try and let you know.

holmesgithub, 2020-04-14:
@PineMaster, Worked like a charm. thanks for the info.

closed 2020-04-14 by holmesgithub