RoganDawes/P4wnP1_aloa #235: No sessions show up (wifi_covert_channel) how to add a language

DonaldDucker34, 2020-04-15:
Well im very new to P4wnP1 and i don't really know how to add an language to the v0.1.0-alpha2. I want to add the German language because i think the language german isn't installed. I am also having trouble with the wifi_covert_channel.js i did everything the same as Seytonic (https://www.youtube.com/watch?v=-QrfCNUxKgE&t=) but when i enter the sessions command no sessions show up and i don't know why :/

PineMaster, 2020-04-15:
dude there is a german format, just have to type layout(de)
https://github.com/RoganDawes/P4wnP1_aloa/blob/master/dist/keymaps/DE_ASCII.json
:V 
I will myself have to test the backdoor thing out, most of these sora get outdated or smth, ill check!

DonaldDucker34, 2020-04-15:
thx for the fast answer. i noticed that even the original Pwnpi (not aloa) does not worked it just opens the program Office and also i dont know why 
(sry for my english its not that great)

DonaldDucker34, 2020-04-17:
as i mentioned the original Pwnpi hidbackdoor also dosnt work i read that you need to make a delay in the Pwnpi.py but i cant find that file on my pi zero

closed 2020-04-27 by DonaldDucker34