RoganDawes/P4wnP1_aloa #247: Windows 10 freezing when it recognizes the device

zerodeaths000, 2020-08-06:
Hi, I recently installed P4wnP1 aloa to my rpi zero w. The problem is, (and was on the first P4wnP1) is that my computer freezes after I hear the standard "usb device connected" sound. What you hear when you plug in a USB Drive. I saw a similar post, and turned off my antivirus(I use Kaspersky) but it still froze. If you have a similar issue, or you can help, feel free to answer. Thanks!

cyb3rr0nn13, 2020-08-06:
Try on another computer

thecoder-001, 2020-08-07:
Hello @zerodeaths000 the problem seems to be growing more and more. Can you please first try on another computer if possible as suggested by cyb3rr0nn13 and also mention your Windows 10 version and build number? Also, can you please provide the driver nave, vendor and version used by your PC for USB devices?
You can try to follow the debug steps as mentioned here: #171 (they can be helpful).
Also, do you use a different language other than English as the default language (input language) for your PC?
Looks like it is really a bug on Windows part and either the driver used for communication is failing or there is something wrong in the hardware management stack for external input devices.

zerodeaths000, 2020-08-07:
Hey! I Will definitely try on another computer! Also, here are the infos:

I used an older Toshiba laptop for it(Satellite L50-B-1K0)
Windows Home 10 Version 1909 Build 18363(.959)
The USB is a UGREEN USB

And yes, im using a different input. Also, I forgot to mention two things:

 - When it happened for the first time, I pressed the completely stopped the pc by the power button. I booted up again(leaving the Pi connected) and it woked fine

- When I RUN one HIDScript then everything fixes and I can now control the system

Also, I'm testing my own language(Hungarian) right now, but for the time when I tested P4wn0P1, I switched to US.

Thanks for your help!

cyb3rr0nn13, 2020-08-07:
Never heard ocf A1B2C3...

thecoder-001, 2020-08-07:
> Never heard ocf A1B2C3...

Yeah, me too. Looks like its rarely seen and happens if you type PIN of your windows many times wrong and serves as a form of captcha before locking you out of the system for some time. I suppose it was triggered because the P4wnP1 aloa booted up before windows and auto triggered an HID script as (maybe) set up by @zerodeaths000 . The P4wnP1 then started to type out whatever was in the script in the PIN dialogue box on the login screen and thus confused windows by reaching max wrong PIN limit.
Also, @zerodeaths000 can you try to use "waitLED(ANY_OR_NONE, 5000)" or (if it doesn't work) "waitLED(ANY_OR_NONE, 10000)" as documented <a href="https://github.com/RoganDawes/P4wnP1_aloa/tree/master#wait-for-led-report"> here </a> by implementing it in your HDI script or through CLI?

NathanielJS1541, 2020-10-08:
@zerodeaths000 By "Turned Off MyKaspersky" do you mean you "Paused Protection" or you actively Exited the MyKaspersky App? I had a similar problem even when Kaspersky was supposedly "paused" - it has to be actually closed. Not offering a fix here - just looking for some clarity in what you stated. I also found that when Kaspersky was the problem, disconnecting the keyboard driver and only having the mouse driver connected would work and allow control of the mouse - could you maybe look at this too?