RoganDawes/P4wnP1_aloa #248: kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa.img shows v0.1.0-alpha2 by MaMe82 in web interface

careyjames, 2020-08-07:
kali-linux-v0.1.1-beta-rpi0w-nexmon-p4wnp1-aloa.img shows v0.1.0-alpha2 by MaMe82 in web interface

does this mean I did not get the newest version loaded correctly?

or is the website not updated on the device?

careyjames, 2020-08-07:
[image]

careyjames, 2020-08-07:
bottom left

careyjames, 2020-08-07:
and how can we tell what firmware we are on?

thecoder-001, 2020-08-09:
Hello @careyjames! 
The web-ui seems to be updated after on the release of v0.1.1-beta under `build/webapp.js`.
[image]
I myself have never observed the version number in the footnote of the website so unfortunately can't tell you right now if it really reflects the latest version as updated in webapp.js.
Also, you can also find the version number of your P4wnP1_aloa by typing `P4wnP1_cli` in the SSH shell of your pi (As documented <a href="https://github.com/RoganDawes/P4wnP1_aloa#21-run-a-keystroke-injection-against-a-host-which-has-p4wnp1-attached-via-usb">here</a>.
It should give a output like this:
````
root@kali:~# P4wnP1_cli 
The CLI client tool could be used to configure P4wnP1 A.L.O.A.
from the command line. The tool relies on RPC so it could be used 
remotely.

Version: v0.1.0-alpha1

Usage:
  P4wnP1_cli [command] ...
````
However, this too seems like a manual variable and could have been forgotten to be updated during the new release. The final and most sure way to check if your P4wnP1 A.L.O.A is updated by going to `/usr/local/P4wnP1/keymaps/` through SSH and checking if the directory contains the following files:
* fi.json
* sv.json
* ch.json
* be.json
* ca.json

To do so, follow these steps:
1. SSH into P4wnP1
2. In the terminal type `cd /usr/local/P4wnP1/keymaps/` (changes working directory) and press enter.
3. Now type `ls` and press enter, it would list all the files in the directory.
4. Verify if the files mentioned above are there in the list you just got.
5. Type `cd` and press enter to switch back to the root directory.

If the files are there, then it means that your P4wnP1 is on the latest version, that is v0.1.1-beta as these keymap layouts were only added in the latest update.