RoganDawes/P4wnP1_aloa #254: Altcodes

Mrreadiness, 2020-08-17:
If you live in a non-English speaking country, you know have a problem with a different keyboard layout. (You can't understanding what the keyboard layout right now). One of the solution for that in Windows it's alt code. It's pressed ALT and from 1 to 3 keys typed in numpad (Google for help). But I don't found how to press numbers in numpad, but in source I saw it like KEY_KP1 for example. May be we have any way to use that?

V0r-T3x, 2020-08-24:
@Mrreadiness 
I already tried to enter ASCII keyboard command, but we can't. We cannot hold a key while we click on other button... Then even you find the right keypad command it's impossible.

[Source](https://github.com/RoganDawes/P4wnP1_aloa) :

-adding multiple keys to the a single press command, doesn't produce a key sequence. All given given keys are pressed at the same time and release at the same time.

-press releases keys automatically, this means a sequence like "hold ALT, press TAB, press TAB, release ALT" currently isn't possible

Mrreadiness, 2020-08-24:
@V0r-T3x!Yes, I know about this problem, but in middle level (C language) programming it's not problem. All keyboard action looks like key pressed and key released. I'm sure it's possible

V0r-T3x, 2020-08-24:
@Mrreadiness  If you find the trick to do this I'm interested to know it. To making a universal keyboard this can be awesome ^^

Mrreadiness, 2020-08-24:
Well, I'll try!

V0r-T3x, 2020-08-24:
I have the same problem, I live in a non-english country too, and I just write a keyboard layout. The file size is 36,6ko, unlikely the others layouts with only 12ko. I know somes characters are useless to make an injection attack, but I love doing things right ^^.

[Canadian Multilingual Standard layout](https://github.com/V0r-T3x/keymaps/blob/master/CMS.json)

And it's not the only one layout in use then if I can merge many layout une only medium size script, this would be great.

Do you have an idea what controls the USB HID command? I curious to make a little research to. There was many more little things I want to upgrade or adjust on the P4wnp1.

Mrreadiness, 2020-08-25:
No ideas yet. I will look for answers in the source code.
What did you mean "I just write a keyboard layout"? You press change language keys? I think it's bad idea because you don't know what the language set right now and how in this PC change keyboard language.

V0r-T3x, 2020-08-25:
@Mrreadiness, no absolutely not. I just custom the common.json in the keymaps. I don't switch between different keyboard layout in the script.

I found this comment in the keyboard.go (line 275 to 295) Do you this it's depend on this? Maybe to manage key hold we have to use different mapping files for rune-mapping and key-mapping.

[Keyboard.go](https://github.com/RoganDawes/P4wnP1_aloa/blob/master/hid/keyboard.go)

// The purpose of mapKeyNameToReport is to resolve the given keyname to A SINGLE REPORT, thus report sequences
// are truncated to the first report only, before returned. This is the trade-off between assuring to return a
// single report per keyname versus managing separated mapping files for rune-mapping and key-mapping.

Mrreadiness, 2020-08-26:
@V0r-T3x can you send example of your common.json, I still don't understand what you are doing with this file. When I have free time, I understand more in the code for the manage key hold.

V0r-T3x, 2020-08-27:
[common.json](https://github.com/RoganDawes/P4wnP1_aloa/blob/master/dist/keymaps/common.json)

[CMS.json](https://github.com/V0r-T3x/keymaps/blob/master/CMS.json)

But this is just my custom keymaps. The file is in /use/local/P4wnP1/keymaps/.

Mrreadiness, 2020-08-27:
Wow, I understand what you are doing, but still don't understand how it works? I tried this on my PC but it doesn't work.

V0r-T3x, 2020-09-09:
@Mrreadiness  i found the logic process of the keysequence there: [Press printscreen in HID? #52](https://github.com/RoganDawes/P4wnP1_aloa/issues/52)

> Well, the press function will call [StringToPressKeycombo](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard.go#L205) internally which will eventually call [mapKeyNameToReport](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard.go#L296) which will eventually look for the keysequence in the loaded [mapping](https://github.com/mame82/P4wnP1_aloa/tree/master/dist/keymaps). However since printscreen is not a character, it won't find anything in there and will look into [keyboard_globals.go](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard_globals.go) (which is where you will also find the other available keys for the press function) and find it there [KEY_SYSRQ](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard_globals.go#L688)
> 
> ### TLDR:
> Try `press("KEY_SYSRQ")` or something similar.

now i'm searching to how the Go language exchange with the usb to see how to add a hold option.

V0r-T3x, 2020-09-23:
@Mrreadiness Finally, i figured out than i can just type that:
`"{": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": ["KEY_KP1", "KEY_KP2","KEY_KP3"]}],`
and I can use the ALT-code. It work well on my PC, I will try with others pc and try different keyboard configuration. I could now try to make a layout who fit on many keyboard layout.

V0r-T3x, 2020-09-26:
@Mrreadiness maybe somethings in this post can help you more to figure how to proceed. @RoganDawes answer somes questions there.

[Full Canadian Multilangual Standard, and the limit of the keymap #257](https://github.com/RoganDawes/P4wnP1_aloa/issues/257)

Mrreadiness, 2020-09-26:
Hi @V0r-T3x!  Thank you very much for your help!  Sorry for the long response time, but I'm very busy right now.  I hope to be back to work on this project next month