RoganDawes/P4wnP1_aloa #257: Full Canadian Multilangual Standard, and limit the limit to keymap

V0r-T3x, 2020-08-23:
I have just a little question to knowing if i can really add all characters to the keymap layout, or if i have to workaround and write a js script in the HIDscript to complete it. 

This is my CMS.json in the /usr/local/P4wnP1/keymaps/:

[CMS.json.log](https://github.com/RoganDawes/P4wnP1_aloa/files/5115136/CMS.json.log)

All is working and i tested it with this HID script:

[HIDscript-testCMS.log](https://github.com/RoganDawes/P4wnP1_aloa/files/5115146/HIDscript-testCMS.log)

But i found some extra special characters to add, but I need two modifiers to achieve this. Its a combination of MOD_RIGHT_CONTROL and MOD_LEFT_SHIFT + the key. I need something like this:

`"¡": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_1"]}],`

Is it possible? if its not, i will write a function to make the command in the injection. Like this:

[HIDscript.log](https://github.com/RoganDawes/P4wnP1_aloa/files/5115137/HIDscript.log)

V0r-T3x, 2020-08-23:
I found how made it. 

`"¡": [{"Modifiers": ["MOD_RIGHT_CONTROL", "MOD_LEFT_SHIFT"], "Keys": ["KEY_1"]}],`

V0r-T3x, 2020-08-24:
@RoganDawes i know someone else worked on a Canadian Multilingual keyboard, but this one was a lack of characters. I extract every keys from the layout and i test each one with a HIDScript. I leave the link to my git files if someone is interested.

This is the exact layout:
[image]

Te code is here:
[CMS.json + CMSTest.js](https://github.com/V0r-T3x/keymaps)

TheMMcOfficial, 2020-09-22:
Good job I will merge this layout to mine and create a pull request! I will try to debug my French Canadian layout at the same time.

V0r-T3x, 2020-09-22:
@TheMMcOfficial excellent ^^. 

Hey by the way, do you know how the code communicates by usb?  I'm trying to add a "hold" function to the script, but I'm still trying to figure out how it works.

RoganDawes, 2020-09-22:
What do you mean by "Hold"?

There is already a `delay()` command in HIDScript, if that is what you are after?

V0r-T3x, 2020-09-22:
No, it's not what I'm searching for. The hold function can give us access to the ALT-code (eventually we can avoid the use of a keyboard layout with it, but the HID controller have to be recoding in some way). 

Ex:
Hold left-alt
Press num_pad_1
Press num_pad_2
Press num_pad_3
Release left-alt

Character: "{"

I'm looking for the part of the code to modify to add a hold fonction. I think that have a relation with the HID controller and the usb report.

RoganDawes, 2020-09-22:
Ah, I see what you mean now.

So, hid/keyboard.go has a function `PressKeySequence`, which emits the provided reports, followed by an empty report to indicate that all keys have been released. This is invoked from hid/controller.go, via function `jsType`, which implements the HIDScript `type()` function. I wonder if the best approach would be to implement a `keySequence()` function that accepts an array of key presses, with no implicit release in between them, only at the end.

e.g.:
```
keySequence("LEFT_ALT NUMPAD_1", "LEFT_ALT NUMPAD_2", "LEFT_ALT NUMPAD_3");
```

Note that the README actually indicates that this is a current limitation:

```
press releases keys automatically, this means a sequence like "hold ALT, press TAB, press TAB, release ALT" currently isn't possible
```

TheMMcOfficial, 2020-09-22:
@V0r-T3x are you able to this this French Canadian/Canadian layout ?
I don't see what is wrong but i'm not able to make it work.

I got error in /var/log/syslog
```
Feb  7 00:37:10 kali P4wnP1_service[149]: START JOB 2 SCRIPT ON VM 0
Feb  7 00:37:10 kali P4wnP1_service[149]: 2020/02/07 00:37:10 HIDScript layout: Setting layout to 'ca'
Feb  7 00:37:10 kali P4wnP1_service[149]: 2020/02/07 00:37:10 HIDScript layout: Couldn't set layout `ca`: Language map with name 'ca' isn't loaded!
Feb  7 00:37:10 kali P4wnP1_service[149]: 2020/02/07 00:37:10 HIDScript type: Typing 'abcdefghijklmno ...' on HID keyboard device '/dev/hidg0'
Feb  7 00:37:25 kali P4wnP1_service[149]: JOB 2 on VM 0 SUCCEEDED WITH RESULT: null
Feb  7 00:37:25 kali P4wnP1_service[149]: GADGET MANAGER HID EVENT: {Type:4 Job:0x87e4780 Vm:0x87e9820 Message:JOB 2 on VM 0 SUCCEEDED WITH RESULT: null}
Feb  7 00:37:25 kali P4wnP1_service[149]: VM 0 set to isWorking: false
Feb  7 00:37:25 kali P4wnP1_service[149]: StartScriptAsBackgroundJob: Job 2 finished or interrupted, removing from global list
Feb  7 00:37:25 kali P4wnP1_service[149]: GADGET MANAGER HID EVENT: {Type:1 Job:0x87e4780 Vm:0x87e9820 Message:Script stopped}
Feb  7 00:37:25 kali P4wnP1_service[149]: Job 2 received IRQ, NOT sending to VM as not attached to any
```

[CA.zip](https://github.com/RoganDawes/P4wnP1_aloa/files/5264536/CA.zip)

V0r-T3x, 2020-09-22:
@TheMMcOfficial i just load your file and i rapidly saw this duplicate and the missing of a mod/key. maybe its just it.

on line 47:
			"°": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
on line 76:
			"°": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
on line 177:
			".": [{"Modifiers": [""],	"Keys": [""]}],

V0r-T3x, 2020-09-23:
@RoganDawes yeah i already saw the note on the limitation, but i was curious to see if we can push the code forward :P. I know its limited to one report, but what is the reason? What part of the code is sending the USB report for the keyboard and what retrieve it? isn't it possible to save the hold key is a certain variable and write the report with a verification of this variable. I'm trying to understand the mechanism behind it.

[USB Human Interface Devices](https://wiki.osdev.org/USB_Human_Interface_Devices)
Is the report similar to the keypress mechanism of this page?

V0r-T3x, 2020-09-23:
@TheMMcOfficial I modify this, and it work.
` 			".": [{"Modifiers": [""],	"Keys": [""]}],`
`			".": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_DOT"]}],`

[CA.zip](https://github.com/RoganDawes/P4wnP1_aloa/files/5264862/CA.zip)

TheMMcOfficial, 2020-09-23:
I will test this later and I will do the pull request thanks a lot in advance.

V0r-T3x, 2020-09-23:
@TheMMcOfficial No problem, it's my pleasure ^^.

RoganDawes, 2020-09-23:
> @RoganDawes You are absolutely right. I think the best approach is with a keySequence. Do you have an idea how to achieve that?

I think the approach that I outlined above would work fairly well. Implement a comparable `jsKeySequence()` function that parses individual arguments exactly the same as `jsPress()` does, packages them up as an array of reports, and passes them to `hid/keyboard.go:PressKeySequence()`

There may still be a requirement to add mappings for the numpad keys, though, I think. I didn't see them when I looked.

TheMMcOfficial, 2020-09-23:
Thanks for your help @V0r-T3x but I still got this error while I run it. P4wnP1 output some thing but I'm pretty sure it's the default and/or the last layout working  that is writing.

`Couldn't set layout `CA`: Language map with name 'CA' isn't loaded!`

V0r-T3x, 2020-09-23:
@TheMMcOfficial What is the HIDscript you use to test your layout? I will try in the same situation then you. And did you change something else in the original source code?

TheMMcOfficial, 2020-09-23:
I got it working
[CA.zip](https://github.com/RoganDawes/P4wnP1_aloa/files/5268248/CA.zip)

RoganDawes, 2020-09-23:
> There may still be a requirement to add mappings for the numpad keys, though, I think. I didn't see them when I looked.

Actually, the keypad keys are mapped using `KEY_KP1`, etc, so that should not require any additional effort.

V0r-T3x, 2020-09-24:
@RoganDawes is there a reason to limit the sequence to 3 keys? Is it because its considerate to 3 pairs, the 6 bytes and with the last null report its more then the 8 max? Because to doing all the alt-code we have to use 4 digit code. Example even the Alt-161(í) differ to Alt-0161(¡).

RoganDawes, 2020-09-25:
No, I was just using the example that you gave. JavaScript can support variable argument length functions, and I'd hope that that could be mapped fairly sanely into Go, perhaps just as a `[]string`. Will have to check to see exactly how Otto (the JS interpreter) would do that, otherwise it might need some sort of wrapper function in JavaScript that takes a variable number of arguments, checks that each one is indeed a string, converts that to an array of String, and then invokes the Go function.

RoganDawes, 2020-09-25:
In truth, we'd probably be better served by just implementing `hold()` and `release()` functions, but I'm not sure how best to maintain the state across multiple underlying function calls. Perhaps I should take a look at that!

V0r-T3x, 2020-09-26:
@RoganDawes what and how, the report is sent? is the sent report  have to look something like this keypress mechanic?

[Keypress mechanism](https://wiki.osdev.org/USB_Human_Interface_Devices)

```
Alt-0161 = ¡

-hold alt-
Alt-0
04 00 62 00 00 00 00 00

Alt-1
04 00 59 00 00 00 00 00

Alt-6
04 00 5e 00 00 00 00 00

Alt-1
04 00 59 00 00 00 00 00

Release
00 00 00 00 00 00 00 00
```

RoganDawes, 2020-09-27:
Yes, that is correct. As mentioned above, `hid/keyboard.go:PressKeySequence()` accepts an array of reports (where a report is essentially the sequence of 8 bytes that you showed) to emit (excluding the final "nil report"). It emits the individual reports by writing them to the `/dev/hidg*` device, then finishes by writing a final "nil report" to ensure that there are no more keys pressed. 

It may be worth extracting the code that does the writing to the device, and making it available to other methods that could maintain a "state" of held modifier keys, as well as improving the implementation of timing-related functionality. (As seen at https://github.com/RoganDawes/P4wnP1_aloa/blob/master/hid/keyboard.go#L401, there is no delay between the individual reports in the sequence, just between the reports and the nil report).

V0r-T3x, 2020-09-28:
@RoganDawes  I'm trying to think some manner to add the hold function. I don't know if they are good ideas or if it's optimal.

Keyboard_layout:
```
suggestion 1:
If the entry have more then one "Keys", Modifiers are hold.

"¡": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": ["KEY_KP0", "KEY_KP1", "KEY_KP6", "KEY_KP1"]}],

Alt+0
Alt+1
Alt+6
Alt+1

suggestion 2:
(I don't know if this variation is possible)
If the entry have more then one "Keys", modifiers are on hold, and each "..." are combine with the hold modifier.

"¡": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": ["KEY_A KEY_B", "KEY_C KEY_D KEY_E"]}],

Alt+A+B
ALT+C+D+E
```
Function:
```
suggestion 1:
holdPress("MODIFIERS_TO_HOLD", "KEYS_SEQUENTIAL");
holdPress("LEFT_ALT", "KEY_KP0 KEY_KP1 KEY_KP6 KEY_KP1");

Alt+0
Alt+1
Alt+6
Alt+1

suggestion 2:
more complex combination:
holdPress("KEYS_TO_HOLD", "1st_KEYS_TO_COMBINE", "2nd_KEYS_TO_COMBINE", "...");
holdPress("KEY_X KEY_Y", "KEY_A KEY_B", "KEY_C KEY_D", "KEY_E KEY_F");

X+Y+A+B
X+Y+C+D
X+Y+E+F
```

RoganDawes, 2020-09-28:
Do you see a problem with the `keySequence` approach I suggested? That does exactly what you are suggesting, without the complicated syntax, or potentially confusing semantics.

V0r-T3x, 2020-09-28:
Ok I see. Yeah you are right, it's a bit longer but it's much more versatile. If we can use it inside the keyboard layout too, it's perfect. I have nothing more to say.

V0r-T3x, 2020-10-12:
@RoganDawes  Did you have the time to continued the development of the coding?

Taino23, 2020-10-12:
I have been looking to start

Sent from ProtonMail Mobile

RoganDawes, 2020-10-13:
> @RoganDawes Did you have the time to continued the development of the coding?

I thought perhaps you were going to have a go?

V0r-T3x, 2020-10-14:
@RoganDawes I'm still learning the language :sweat_smile: . I can make keyboard layout by deduction or HID script, but I don't have the knowledge at this time to make it.  I even have problem to compile the script with build.sh.... :sweat_smile: .

closed 2025-01-03 by V0r-T3x