RoganDawes/P4wnP1_aloa #260: Settings reset after refreshing

doge-co, 2020-08-27:
Hi,

After I store my settings and refresh the page they are reset to factory settings why is this?

V0r-T3x, 2020-08-28:
Its normal if you refresh the page all your stored settings will be reset. You have to deploy your new settings to active it.

doge-co, 2020-08-28:
It still happens after I deploy the settings, I get this error message too: Error while deploying new USB gadget settings
rpc error: code = Unknown desc = Response closed without headers.

V0r-T3x, 2020-08-28:
What do you try to deploy? I already had one error like this when I tried to deploy a keyboard, a mouse AND a mass storage. If I try just the keyboard and mouse, this work. If I try just keyboard and mass storage, this work too. But three of them don't work.

doge-co, 2020-08-28:
I just tried deploying then saving the settings and it worked.

doge-co, 2020-08-28:
It was keyboard and mass storage.

closed 2020-08-28 by doge-co