RoganDawes/P4wnP1_aloa #269: wifi connection disappears after update

sarsmen, 2020-09-20:
updated metasploit after updating packages
sudo apt update && apt upgrade the wifi connection is lost or it is connected but the ip address is not issued 172.24.0.1 it is issued completely different.

01- apt install ruby-dev

02- cd /usr/share/

03- git clone https://github.com/rapid7/metasploit-framework.git

04- cd metasploit-framework

05- apt install git-core postgresql curl nmap gem libsqlite3-dev

06- gem install wirble sqlite3 bundler

07- gem install nokogiri

08- gem install bundle

09- bundle install

NYPDK, 2020-09-26:
Yeah, after doing sudo apt-get upgrade I have the same issue...

sarsmen, 2020-09-28:
Don't know what i did. after the update, the third time it turned out.

V0r-T3x, 2020-09-30:
Do you try `sudo dhclient` command?

NeuronAddict, 2020-11-07:
Same issue when upgrade and connected to USB port of raspberry pi W (no power).

When make apt dist-upgrade, network-manager switch to the usb connection when upgrade. The side effect is the lost of the wifi (and of course the ssh terminal when the update is perform).

I was avoid this by :
 - Use power USB port only when upgrade
 - Use a wifi template with client + AP or client.

Hope this help.