RoganDawes/P4wnP1_aloa #278: pi0 0w rpc error code = unknown desc = Responce closed without hearers

socket23, 2020-10-31:
I was just terying to setup a a starrtup template so that i could access kali or the settings gui via Wifi instead of having to plug in via USB A. I did something and when i deploy any new setting now I get the error in the title. any suggestions??

michaelpineirocode, 2020-11-22:
I also have this same issue. It is aggravating. If anybody else has any input, I would appreciate it!

michaelpineirocode, 2020-11-27:
I finally figured it out, if anybody needs help! Basically, instead of immediately deploying anything, store it as a setting, and then add it to a template in the "generic settings" tab. Then, you can deploy the template with all the changes you made. Good luck!

socket23, 2020-11-28:
Hey if you want to direct message me the fix or just drop it on here I would love to know

michaelpineirocode, 2020-12-03:
Okay so this is the exact sequence of steps on how I got it to work:
1.) Flash the image onto a micro sd and insert it into the raspberry pi zero
2.) Using the second micro-usb port or by soldering one on, plug the pi into a computer. At this point, you can also power it with a 5V supply by plugging it into a phone charger or something similar, or just power it, in general, using a power cable. Obviously you will need to be plugged into a computer for the actual thing to work, though. I have only tested it on Windows P.C.'s so far.
3.) Wait for the computer to load and then go to wifi networks. Join the P4wnP1 network
4.) Go to a browser and type the IP into the search bar. I noticed it changes sometimes. For me the address was 172.24.0.1:8000
5.) Now, this is where the error normally is. Change the settings to whatever you need, but instead of pressing "deploy," store it as something new. I wouldn't mess much with the network settings except for maybe the name and password if you don't know what you're doing.
6.)After all of your settings are set up, go to "generic settings." Here is where you can load and store templates.
7.) For each section, put your stored settings into it and press "store."
8.) There is also a separate option for the template that will be enabled on startup. Choose the template that you just created and hit "deploy." This will deploy all of the settings that are currently loaded and will save the template for the next time that it boots.

Now, if you want to access "kali," I would ssh into it. Again, power it up and connect to the network, and make sure you have ssh installed. Then, type "ssh root@172.24.0.1 or whatever the IP is. If it connects, type "yes" and then type the password, which is "toor" by default. From there you can type apt-get install kali-linux-full.

I hope that this helped, and best of luck to you or anybody else who keeps running into issues!