RoganDawes/P4wnP1_aloa #279: [...] via raw HID device " Error: "

GiacomoMazza, 2020-11-01:
Hi! I am trying to use the covert wifi feature, but once I type ./wifi_covert_channel.sh I get the following message: Delivering " /usr/local/P4wnP1/legacy/wifi_agent.ps1 " via raw HID device " Error: "

I am not sure what I am doing wrong since I have not changed anything in the code but the keyboard language. Any idea? Thanks in advance!

GiacomoMazza, 2020-11-02:
@mame82 So sorry to mention you directly, but I have spent a week on this and perhaps it's something you might have seen before. Also kinda worried I'll get no replies since previous issues got no replies...

mame82, 2020-11-03:
There is no need to run a script manually.

A MasterTemplate for the WiFi covert channel exists, which sets everything up (bootscripts, proper triggeractions, keyboard script to inject client agent)