RoganDawes/P4wnP1_aloa #281: Stealth mode / login

ToeiRei, 2020-11-26:
I was wondering if it is possible to 'stealth' the web UI a bit making it a bit so it does not scream 'hacker around'; 
It would be nice to have a login page mimicking some innocent web admin to prevent discovery (by non-tech folks) if connected to a wifi network

V0r-T3x, 2021-08-05:
I use a simple solution, but someone with minimal skill can find. I just created a javascript file on the /usr/local/P4wnP1/www/. I used this code:

```
/*p.js*/

var password;

var pass1="toor";

password=prompt('Please enter your password to view this page!',' ');

if (password==pass1)
  alert('Password Correct! Click OK to enter!');
else
   {
    window.location="about:blank";
    }
```

I added `<script src="p.js"></script>` to the index.html page. If the password is wrong, a `about:blank` page is create. You can easily create a page who mimic what you want instead of the blank page. Or maybe, you can merge the content of index.html into an another page an made a fake access portal with a java script code similar to this. Like I said, it's a simple smokescreen and if you look at the page source code someone can easily find it. hope thats can help you.

worthl3ss, 2022-03-31:
> I use a simple solution, but someone with minimal skill can find. I just created a javascript file on the /usr/local/P4wnP1/www/. I used this code:
> 
> ```
> /*p.js*/
> 
> var password;
> 
> var pass1="toor";
> 
> password=prompt('Please enter your password to view this page!',' ');
> 
> if (password==pass1)
>   alert('Password Correct! Click OK to enter!');
> else
>    {
>     window.location="about:blank";
>     }
> ```
> 
> I added `<script src="p.js"></script>` to the index.html page. If the password is wrong, a `about:blank` page is create. You can easily create a page who mimic what you want instead of the blank page. Or maybe, you can merge the content of index.html into an another page an made a fake access portal with a java script code similar to this. Like I said, it's a simple smokescreen and if you look at the page source code someone can easily find it. hope thats can help you.
> 
> edit: I think to it, we can add a little code to retrieve information about the connection try. This script can take the timestamp, ip and the password attempt.

You'd ideally hash the password to avoid timing based attacks one could do