RoganDawes/P4wnP1_aloa #285: Wifi Settings are not being saved

chiragvshroff, 2020-12-26:
Onec i change the ssid and password  i hit deploy it applies the new settings and works fine but after a reboot the wifi settings go back to the default one

Shevnet, 2020-12-27:
I'm having the same issues. Anyone have any ideas on a resolution?

21weberer, 2020-12-27:
You need to actually save the wifi settings, not just deploy them.

chiragvshroff, 2020-12-27:
> You need to actually save the wifi settings, not just deploy them.
> 
> 
> > I'm having the same issues. Anyone have any ideas on a resolution?

Can you please explain it in steps please ?cause I guess I have saved it in the wrong method ￼

21weberer, 2020-12-27:
> #285 (comment)

After making everything on the screen, click the store button, located next
to load stored. Make up a random name, or something like that, then save.

21weberer, 2020-12-27:
Another thing, you need to set the master template to always load your new
config file

chiragvshroff, 2020-12-27:
> Another thing, you need to set the master template to always load your new
> config file
> 
> 
> > You need to actually save the wifi settings, not just deploy them.
> >
> >
> > I'm having the same issues. Anyone have any ideas on a resolution?
> >
> > #285 (comment)
> > ,
> >
> >
> > Can you please explain it in steps please ?cause I guess I have saved it
> > in the wrong method ￼

Thanks a lot now it works perfectly I had not set the master template, also do you know how to delete partitions created by using genimg?￼￼

21weberer, 2020-12-29:
Yes, you can, but you need to be able to ssh into the pi, and from there
you can find the folder that contains the partitions, and delete the image.
I do not remember the exact path.

closed 2021-01-05 by chiragvshroff