RoganDawes/P4wnP1_aloa #302: My p4wnp1_aloa wifi is not detected in wifi list

hackingwhangjuncomputer, 2021-08-15:


hackingwhangjuncomputer, 2021-08-15:
[image]

It's obviously working...
[image]
[image]

hackingwhangjuncomputer, 2021-08-15:
[image][image]

hackingwhangjuncomputer, 2021-08-15:
I plugged it into my MacBook, but it only says No connection to network settings and I can't get wifi either.

TheMMcOfficial, 2021-08-16:
Check your cable. I had problems with one of my old cable. The raspberry pi was rebooting so the wifi had no or almost no time to show up. I bought some better one on amazon and everything was working after.

hackingwhangjuncomputer, 2021-08-16:
이 캐이블은 3일전만해도 작동했습니다

hackingwhangjuncomputer, 2021-08-16:
This cable worked 3 days ago.

ipse-diabolus, 2022-02-04:
Hi,

I had a problem with channel 13 2.4 GHz, which was not fixed even by changing the region code.

qkum, 2022-04-04:
The name of the default wifi contains emojis and bugs out on kali linux. 
Likely the same on mac.
But works on windows 10.

qkum, 2022-04-04:
> This cable worked 3 days ago.

It can break at any time if you are unlucky. (bending it)

It is just the copper wires that need to break inside and the cable will be defective. 

Anything is possible. 

Your Mac could also have its hard disk corrupt today etc.