RoganDawes/P4wnP1_aloa #307: Raspberry Pi Zero 2 W Support

georgewoodall82, 2021-10-30:
I have a Raspberry Pi 0 2 W and it does not boot. Any Future plans to support it?

whackyhack, 2021-10-31:
This project is mostly Go.  I suspect that you can build yourself from live KALI Linux.

C-Noob404, 2021-11-08:
That would be great! If you know a solution I would be very gratefull for the instructions.

Brazier85, 2021-11-08:
I tried to use this script here: https://gitlab.com/kalilinux/build-scripts/kali-arm/-/blob/master/rpi0w-p4wnp1-aloa.sh

But the generated images did not boot.

italoaa, 2021-11-08:
same issue but no luck. Thank you anyways for saying it does not support Pi z 2

YourFriendCaspian, 2021-11-10:
I too would be very grateful for instructions on how to get this running on the Zero 2. Any help would be greatly appreciated.

C-Noob404, 2021-11-10:
I got the img that this script produces to boot. [https://gitlab.com/kalilinux/build-scripts/kali-arm/-/blob/master/rpi0w-p4wnp1-aloa.sh](url)
But I have no idea how to connect to it. Changing the "wpa_supplicant.conf" before boot doesn't do anything.

JamesC00, 2021-11-11:
"At the heart of Raspberry Pi Zero 2 W is RP3A0, a custom-built system-in-package designed by Raspberry Pi in the UK"
so i have been using rpi3 images. it seams to work. i'll see if i can get p4wnp1 working on mine tonight

clout86, 2021-11-11:
I do have an urge to try and port this package to an ebuild for Gentoo
linux, perhaps with the Pentoo overlay, in which case would make porting to
other system architecture  simpler.   Albeit, as long as they support USB
gadget mode or the likes. It may better suffice being a portage Overlay on
it's own as it seems to be more of a distro with a custom init rather than
a simple package that is installed. Skimming that boot-strap script:
`https://gitlab.com/kalilinux/build-scripts/kali-arm/-/blob/master/rpi0w-p4wnp1-aloa.sh
<http://url>` and it'd definitely beyond the point of being a simple
package.

Cheers

tomo-one, 2021-11-20:
Older versions of Kali do not boot (at least up to 2020.4). Kali 2021.3 works - at least it boots. Have to check if the script mentioned by @clout86 and @C-Noob404 will work as well.

blitz9826, 2021-12-05:
I recompiled the script with making changes to the architecture, switching armel to armhf. It wasn't very easy, and somehow got around to booting, but I can't use USB, ethernet doesn't register as a result, and there is no SSID broadcasting. Some steps I did:

added stretch repo for libesd0-dev
used https:// instead of git:// for linaro cloning
sudo apt-get install libc6-armel-cross libc6-dev-armel-cross binutils-arm-linux-gnueabi gcc-arm-linux-gnueabi bc qpdf kpartx systemd-container lib32ncurses6 

updated rpi0w-p4wnp1-aloa.sh by replacing all instances of 'armel' to 'armhf'

ran the script: ./rpi0w-p4wnp1-aloa.sh 2021.4

I'm looking into any potential kernel differences by looking at the RPi02W script, but not having any luck.

JamesC00, 2021-12-10:
so yes the wifi chipset is diff and you need modded firmware to run mon mode. as far as i know. but the kali image for the rp z 2w  so i'll try it now. and see what i can get

blitz9826, 2021-12-10:
More progress made: flashed the publicly available ALOA image onto flash, then replaced the boot partition contents entirely (deleted everything then copied) the contents of the compiled Kali Pi Zero 2 image's boot partition. This got me USB but not anything else. To get ethernet working (both WaveShare hat RTL chip and a TP-Link adapter), I copied the /usr/lib/modules contents for the new kernel and got that working. I can SSH into the OS now, dmesg indicates a few errors, and P4wnP1_cli responds to my queries, so I've made decent progress. 

Problems:

BT doesn't work
WiFi doesn't work
OTG doesn't seem to work(?)

Apt update is still using armel arch to install new binaries. Hopefully this goes through, then I'll get pciutils and see if the WiFi adapter is detecting correctly and how else I can get it running (though the lack of nexmon might hurt?).

Alternative approach I'll begin in parallel is to compile the older P4wnp1 directly on the new Kali image and see if I can get at least basic P4wnp1 functions. Cheers!

Compr0mzd, 2021-12-11:
Can't wait to see any updated to this ! I'm so deseperate to create my own unpractical HID

NotReeceHarris, 2021-12-19:
I have a working Kali + aloa image just need to get HID attacks working but after that I will create a pull request with the new source code

V0r-T3x, 2021-12-30:
@NotReeceHarris Do you have notes about yours steps to achieved this? I have a working img of kali pitail for pi02, but I had some problem to install p4wnp1. 

Cant wait for it and I know two guy who cant wait to work on the P4wnP1 on rpi02  @beboxos @p4wnsolo

p4wnsolo, 2021-12-30:
Agree
Id love to have any working semblance of p4wnp1 + pi02

Compr0mzd, 2022-01-06:
Any updated from anyone here ?

lgeekj, 2022-01-09:
I managed to get a booting and "almost" full working P4wnP1-aloa on zero 2w. 
I used the kali image, for the 2w, and basically followed the Makefile from this repo. 
Except when it came to golang, i used the golang from kali repo. (apt-get install golang-1.17)
HID, Ethernet works.
[https://i.imgur.com/MAfkr7p.gif](url)
Rest can't test, because it's much easier to do via web interface .

Now i can't get is the web interface fully loaded.
this problem. [https://i.imgur.com/KT4tJrB.png](url)

lgeekj, 2022-01-09:
Update. 
I forked P4wnP1 projet. Mainly its just changes on Makefile and adding webapp.js to root directory...
Setup wifi normally on kali (kalipi-config)
install git
https://github.com/lgeekj/P4wnP1 -> just clone it on pi zero. 
and run "make dep / make compile / sudo make installkali"
and reboot

V0r-T3x, 2022-01-10:
Thanks @lgeekj I tried it an it seems to work. I'm starting to test it and Ill give feedback if I see something.

RoganDawes, 2022-01-10:
Thanks @lgeekj. I don't have a Pi Zero W 2, so am unable to test your changes. If you would like them to be included in this repo, please consider squashing your commits, as I see a large number of changes, that are a single character, or an error, then fixing that error.

semchapeu, 2022-01-11:
@lgeekj I tried installing your version, however during "make compile" the compilation of /service/datastore failed. The only thing I did differently to your instruction is upgrade kali (apt full-upgrade) beforehand.

lgeekj, 2022-01-11:
> @lgeekj I tried installing your version, however during "make compile" the compilation of /service/datastore failed. The only thing I did differently to your instruction is upgrade kali (apt full-upgrade) beforehand.

Yes, make compile will output an error. Don't know fix, nonetheless if you continue it will work.
Sure, full upgrade should be fine, just that stock Kali, comes full of packages that take a while to update.

> Thanks @lgeekj. I don't have a Pi Zero W 2, so am unable to test your changes. If you would like them to be included in this repo, please consider squashing your commits, as I see a large number of changes, that are a single character, or an error, then fixing that error.
Sure, would like to. Will first, clean up the makefile a bit, also it's needed to add webapp.js (will add wget of that from my fork)

V0r-T3x, 2022-01-12:
@lgeekj I made an img based on your git. 
there's my [notes](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/notes_PIZ2%2BKali%2BP4wnP1%2BLCD1.3)
The first problem I see is the bluetooth don't work, but I don't see if it's related with kali (I haven't verify if it work just with kali first) or if I have to install bluez (or other things). Is the bluetooth is working with your tests?

lgeekj, 2022-01-12:
> @lgeekj I made an img based on your git. there's my [notes](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/notes_PIZ2%2BKali%2BP4wnP1%2BLCD1.3) The first problem I see is the bluetooth [...]

Yes, bluetooth doesn't work. From what I tested, it's from Kali. Doesn't show on "rfkill list" as phy (physical interface) or on lshw.
I also am busy with stuff from university... 
But would like to use Kali arm scripts, to build an image without GUI and some other to make it lighter...
How did you build an <4GB image?

V0r-T3x, 2022-01-12:
@lgeekj ok I will search if theres a way to make the bluetooth work. 

The <4gb img is just the result of the compression of my img, after the 'apt update' and the installation of all package for LCD, the img is in fact ~12gb (I will edit in my note, to avoid misunderstanding)

V0r-T3x, 2022-01-15:
@lgeek, @beboxos flagged another problem about the GPIO on the img I shared. Do you have similar problems too? He asked for tips to @mame82 on twitter. If you would see the thread it's [here](https://twitter.com/BeBoXoS/status/1482115934206758915).

lgeek, 2022-01-15:
You want @lgeekj here, not me :)

V0r-T3x, 2022-01-15:
@lgeek sorry '^-^ haha

teospl, 2022-01-15:
First of all thank you guys for trying to make it work on pi zero 2w. 

Im using current kali for this device.
I flashed image with Pi Imager.

During make dep I was getting golang error saying that go get cant install packages in current version (1.17) so i replaced all go get -option entires in Makefile
to:
go get -d package
go install package

and make dep finished without errors. Then make compile finishes with errors. Then installkali goes ok.
And after reboot I can't access PI via wifi, but I can get to P4wnP1 aloa web panel with usb cable and ethernet (pi is recogized as ethernet card)
Unfortunnetly no commands are being triggered from web panel, I can't configure access via wifi and so on... it looks like system it's up, but most of the features don't work for me. 
in ssh console I'm getting phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52 

Does anyone have any idea what am I doing wrong?
Also, I accidently turned ethernet card off in P4wnp1 panel and I can't turn it back on now

V0r-T3x, 2022-01-15:
@teospl , I didn't change anything from the @lgeekj make file and I didn't got error while make dep. We all got error while the make compile phase. I save my [notes](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/notes_PIZ2+Kali+P4wnP1+LCD1.3) and the [log](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/P4wnP1-rpiZ2.log) I got if it can help you. 

At this time
working for me:
- C/AP wifi
- keymap / HID script
- mass storage

not working for me:
- Bluetooth
- Wifi covert and all which need monitor mode
- Input/output on GPIO

It's a huge WIP and some features depend of the future nexmon firmware for the rpi02w

LegendL3n, 2022-01-16:
@V0r-T3x have you ever got bluetooth working on this Pi?
I'm using the Kali image and can't get it to work as well :\

V0r-T3x, 2022-01-16:
@LegendL3n Saddle not actually. I didn't verify the bluetooth state before installing P4wnP1 on kali on the rpi02w, then idk if it's a firmware problem or a bad config. Yesterday, I used my time trying to figure out what is going wrong with the onboard led and GPIO control, then I start over with my kali (no-tail) img. Today, I will take a look to the bluetooth too.

Evilran, 2022-01-23:
What's the status now? Any updated from anyone here?

V0r-T3x, 2022-01-24:
@Evilran no new progression on my side, Im lacking of time. I search for GPIO problem, but didnt found the answer for it, and this putting some dev in pause. Like I said, Idk if the bluetooth dont work cause the firmware or config. If you can clarify the GPIO problem this will help the project.

Evilran, 2022-01-25:
@V0r-T3x I installed P4wnP1 on kali according to the instructions in your notes, and I had the same problem you described above. Although there were some problems, at least it started successfully. I'll try to fix the GPIO issue (it's firmware related, I guess).

Non808, 2022-02-02:
@lgeek tried to run the `make dep` command on my pi 2 zero w but it failed at:
```
go get -u github.com/gopherjs/gopherjs
go get: installing executables with 'go get' in module mode is deprecated.
	To adjust and download dependencies of the current module, use 'go get -d'.
	To install using requirements of the current module, use 'go install'.
	To install ignoring the current module, use 'go install' with a version,
	like 'go install example.com/cmd@latest'.
	For more information, see https://golang.org/doc/go-get-install-deprecation
	or run 'go help get' or 'go help install'.
# github.com/gopherjs/gopherjs
panic: runtime error: slice bounds out of range [:10] with capacity 8

goroutine 1 [running]:
cmd/link/internal/ld.decodetypeName(0x1174000, 0x740a, 0x108d460, 0x40)
	/usr/lib/go-1.17/src/cmd/link/internal/ld/decodesym.go:131 +0x124
cmd/link/internal/ld.(*deadcodePass).decodeIfaceMethod(0x108d5bc, 0x1174000, 0x415f80, 0x740a, 0x40)
	/usr/lib/go-1.17/src/cmd/link/internal/ld/deadcode.go:418 +0xa4
cmd/link/internal/ld.(*deadcodePass).flood(0x108d5bc)
	/usr/lib/go-1.17/src/cmd/link/internal/ld/deadcode.go:192 +0x9f8
cmd/link/internal/ld.deadcode(0x115a000)
	/usr/lib/go-1.17/src/cmd/link/internal/ld/deadcode.go:330 +0x4c
cmd/link/internal/ld.Main(0x415f80, {0x4, 0x8, 0x1, 0xd, 0xe, 0x1c00000, {0x0, 0x0}, {0x295194, ...}, ...})
	/usr/lib/go-1.17/src/cmd/link/internal/ld/main.go:252 +0x11d0
main.main()
	/usr/lib/go-1.17/src/cmd/link/main.go:69 +0xbb0
make: *** [Makefile:30: dep] Error 2
```

No idea where to start looking to fix this. any ideas?

V0r-T3x, 2022-02-03:
@Non808 do you use the kali (non-tail) image 2021.4 and did you upgraded it?

I'm trying to install it directly on kali non tail 2021.4 image not upgraded without root perm and I got an other error:
```
webaps.js: Permission denied
make: *** [Makefile:26: dep] Error 1
```
I retry it with the root permission for dep/installkali but not for compile and this goes without error. 

Theres my [log](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/logs/P4wnP1-rpiZ2-2022-02-02.log)

Give us more detail, I cant reproduce your error for now.

Non808, 2022-02-03:
@V0r-T3x Had upgraded and didn't do it on root. Did those two compiler errored, but install worked overall (I think it was bluetooth that was broken). Thanks of the advice!

HomeNager, 2022-02-22:
Did you activated bluetooth in /boot/config.txt?
dtparam=bluetooth=on did it for me.

C-Noob404, 2022-02-22:
Unfortunately HID is not working for me. It just don’t write anything. Not even the sample script.

V0r-T3x, 2022-02-23:
@HomeNager I tried you fix technic and it doesn't work for me. I tried it with my img and the @lgeekjopt img. 
I added `dtparam=bluetooth=on` to the `/boot/config.txt` file, rebooted.
I started the bluetooth service with `systemctl start bluetooth`.
but when I tried the `bluetoothctl` command I always finish with the `No default controller available`.

did you do an other things to make it work?

V0r-T3x, 2022-02-23:
@lgeekjopt I just tried your img and @C-Noob404 seems right about the HID not working. The mass storage isn't working too. The usb ethernet connect is working, I can connect on `172.16.0.1:8000` and the CDC ECM is recognized by win10 at least.

V0r-T3x, 2022-02-23:
@C-Noob404 the img of @lgeekjopt has no GUI. I think all the web interface is broken, at least the HID and the mass storage. If you use the `P4wnP1_cli hid [cmd]` you can run HID scripts. I shared an bigger but slower img higher in this thread.

lgeekjopt, 2022-02-23:
Hello, i just tested my img, downloaded from github. And flashed on sd card. 
It boots normally, and creates hotspot (and webinterface works 172.16.0.1:8000)
HID script works, just have to "load&replace" for other (since my Keyboard layout is PT, i use the demo_thumbsup.js created by me), hence the layout("'pt").

For the mass storage, i first expand file system via raspi-config -> Advanced...
next generate a .bin for usage in P4wnP1 "/usr/local/P4wnP1/helper/genimg -s 2048 -l ducky -o ducky" this creates a ducky.bin that emulates usb storage with 2GB and label ducky. 
I uncheck the "CDC ECM" (since i test in windows), and enable the mass storage. 
I link a gif that i made [https://i.imgur.com/EWdftnQ.mp4](url)

soutzis, 2022-03-27:
Nexmon now officially supports Raspberry Pi Zero 2 W (see https://github.com/seemoo-lab/nexmon/pull/514).

Does this mean that it will be possible to fully support PwnP1 A.L.O.A on RPI Zero 2 W ?

DrSchottky, 2022-04-02:
> Nexmon now officially supports Raspberry Pi Zero 2 W (see [seemoo-lab/nexmon#514](https://github.com/seemoo-lab/nexmon/pull/514)).
> 
> Does this mean that it will be possible to fully support PwnP1 A.L.O.A on RPI Zero 2 W ?

Afaik P4wnP1 relies on a [custom wifi fw](https://github.com/RoganDawes/nexmon_wifi_covert_channel) (based on nexmon) since it does HW offloading for stuff like KARMA.
Even if Nexmon is now supporting 2 W there's still quite a lot of firmware code that needs to be ported to get it working.

TDBJR, 2022-04-04:
> Hello, i just tested my img, downloaded from github. And flashed on sd card. (This only for PiZero2W) It boots normally, and creates hotspot (and webinterface works) 172.16.0.1:8000 <- usb/ethernet or 172.24.0.1:8000 <- wifi HID script works, just have to "load&replace" for other (since my Keyboard layout is PT, i use the demo_thumbsup.js created by me), hence the layout("'pt").
> 
> For the mass storage, i first expand file system via raspi-config -> Advanced... next generate a .bin for usage in P4wnP1 "/usr/local/P4wnP1/helper/genimg -s 2048 -l ducky -o ducky" this creates a ducky.bin that emulates usb storage with 2GB and label ducky. I uncheck the "CDC ECM" (since i test in windows), and enable the mass storage. I link a gif that i made [https://i.imgur.com/EWdftnQ.mp4](url)
> 
> related to bluetooth, the recommendation of @HomeNager didnt work. AFAIK, the bluetooth issue is from Kali itself. It was disclosed on they're recent update/changelog.

I'm unable to ssh in with kali:kali. 

debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
debug1: read_passphrase: can't open /dev/tty: No such file or directory

V0r-T3x, 2022-05-17:
I'm trying to make a new img on kali 2022.2. Actually the GPIO don't work wet but the bluetooth work partially. I can connect with ssh by BT. Usb keyboard, mouse, mass storage, wifi, BT, badusb seem working. A small step further.

lgeekjopt, 2022-05-17:
Me too. I also have got Bluetooth working. At least showing up. 
Adding "dtparam=krnbt=on" at end /boot/config.txt

But didn't get much utility from it. How you get shell (ssh) via BT?
No progress on the GPIO. Complex matter, golang, libraries. etc.

V0r-T3x, 2022-05-17:
@lgeekjopt when I saw the BT showing up, I activated connectable, discoverable, bondable and secure simple pairing (and NAP was already selected) but my first attempt failed. To see if it was possible to connect directly, I used `bluetoothctl`, `scan on`, `trust *mac*`, 'pair *mac*`, `connect *mac*`. I activated the bluetooth network on my cell and closed the wifi network. I connected on 172.26.0.1 on juice ssh. After this first connection the BT link is stable and pretty fast.

V0r-T3x, 2022-05-19:
Just thinking at loud. 
The GPIO seems to depend mostly on the periph.io lib. The GPIO don't have problems with kali, I can link button from my hat with a keyboard key on the /boot/config.txt, and it work, I verified with wiringpi too.

@lgeekjopt, I saw you updated the version of golang to the current version. Is the 1.10 version caused problems on compile? And did you already saw the [Kali build script](https://gitlab.com/kalilinux/build-scripts/kali-arm/blob/ab7f4f34d22ec5d383c2afb3ab57b1861ff4a635/rpi0w-nexmon-p4wnp1-aloa.sh)  for the p4wnp1?

V0r-T3x, 2022-05-30:
I tried to update the path source for the periph.io like suggested into the issue i opened. This didn't have corrected the link for the triggers of the P4wnP1, but I tried the `gpio-list` from `https://github.com/periph/cmd` and I can interact with the gpio pins.

@RoganDawes , is the badger module related with the datastore is needed to interect with the periph.io subsys? If not, what put the periph.io in relation with the web ui?

Another anomaly i found is:
P4wnP1_cli led -b 0 is working but instead 0=OFF and >254=ON, it is the opposite, 0=ON and >254=OFF.

I made another file with [my notes](https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/P4wn1_aloa-rpi02w-v2.md)  from this test.

RoganDawes, 2022-05-31:
To be honest, I'm not really that familiar with the P4wnP1 code base, primarily being a host rather than the author (MaMe82 handed it over to me). I also don't have a Pi Zero 2 to test with.

That said, from an electrical perspective, quite often signals are inverted and are "active low", which might explain what you are seeing. Eg the led is directly connected to 3v3, and to the gpio, so bringing the io high is "off", and low is "on".

MatthieuOByrne, 2024-02-10:
Hello, I know I am a little late but does anyone have a working image?
@V0r-T3x @RoganDawes ,

RoganDawes, 2024-02-10:
Sorry, I do not own a 2W.

V0r-T3x, 2024-02-10:
@MatthieuOByrne no I don't have a working image, in fact at the time we didn't succeed in installing it on the rpi02w, there was a problem with GPIO and bluetooth. I sadly lost contact with the other people working on it.

blitz9826, 2024-02-10:
Last success I saw (and happy to try again) was someone’s fork of ALOA
using Kali to start with and adding ALOA after the fact. No BT and GPIO as
mentioned earlier but it got a fair bit of it going.

MatthieuOByrne, 2024-02-10:
I am getting my Rpi z 2 on monday and I am going to try and run [this one](https://github.com/lgeekjopt/P4wnP1_aloa/releases/tag/pi_zero2w).

V0r-T3x, 2024-02-10:
I still have my old notes about the p4wnp1 on github.

https://github.com/V0r-T3x/My_Notes_on_P4wnP1/blob/main/P4wn1_aloa-rpi02w.md

I don't know if this can help you.

Recently I reopened the p4wnp1 code cause I'm planning to add some new similar features to a pwnagotchi XD, it's still just an idea.

But I have to confess the p4wnp1 is missing me.

[10 more replies]