RoganDawes/P4wnP1_aloa #309: System Kernel Modules Not Loading

xi-pinmping, 2022-01-18:
I’ve tried with multiple cables and multiple installs but can never get my WiFi interface to be visible, after connecting the pi to a tv (ssh is broken as well, even over USB), I can log into kali and systemctl is stating that P4wnp1.service, and system-kernel-modules.service are failing and exiting almost immediately. I have followed multiple setup guides and edited config.txt and cmdline appropriately (they were already edited but I did it after multiple failures to be safe).

qkum, 2022-06-02:
> I’ve tried with multiple cables and multiple installs but can never get my WiFi interface to be visible, after connecting the pi to a tv (ssh is broken as well, even over USB), I can log into kali and systemctl is stating that P4wnp1.service, and system-kernel-modules.service are failing and exiting almost immediately. I have followed multiple setup guides and edited config.txt and cmdline appropriately (they were already edited but I did it after multiple failures to be safe).
> 
>` Nothing changes this behavior. Please get back to me on this issue.`

Anything to add now months later?

xi-pinmping, 2022-06-13:
I haven’t had time to take a look at it as I’ve been buried in other projects but I have made no progress since the last post.