RoganDawes/P4wnP1_aloa #310: Please add support for Pi 4

xi-pinmping, 2022-01-24:
Hi, I’ve been a big fan of this project for a while but now that it’s had some age it seems not to want to work on the Pi 4. 

I’ve been playing around with it for a week or so now and was unable to get it to run on the new model, have tried installing kali for arm and mimicking the setup (as a USB Ethernet device), but I can’t get the pi to connect. I have tried setting it up in gadget mode to no avail. 

Any assistance is appreciated, and if I can get it working I’d be more than willing to contribute.

whackyhack, 2022-01-24:
Much of P4wnP1_aloa hangs on USB-on-the-go (USB OTG) which is not in any of Raspberry Pi classic series.  In short, the USB functions of P4wnP1_aloa (simulating a USB client, i.e., being a USB gadget) will not work on Pi 4.

clout86, 2022-01-24:
I believe the pi 4b board can support the USB gadget mode of the linux
kernel, though I have not tested this.

xi-pinmping, 2022-01-24:
I have seen some say they had success getting gadget mode to work with kali. It is supported on the pi 4, but not by default. There’s a couple articles about it I’ve been reading but following them yields a return that doesn’t work. I can boot to kali after it’s plugged in but the script to get a network connection fails and on my host machine (Arch) I see a “connection to interface *long Ethernet name* failed” notification every minute or so.

qkum, 2022-04-07:
> Hi, I’ve been a big fan of this project for a while but now that it’s had some age it seems not to want to work on the Pi 4.
>
> I’ve been playing around with it for a week or so now and was unable to [...]

It is so buggy as it is that it does not make sense to port it to RPI4 before fixing those bugs.
- Fixing the current bugs should be easier than porting to RPI4.
- You will just get those bugs on the RPI4 if they are not fixed first.

JollyJamma, 2023-08-08:
A Pi Zero W is £15/$15. That's not a lot of money for a device that is natively supported by this project. 

Get that.

Besix2, 2023-08-08:
Yeah if you could buy the pi zero for that price everything would be ok. Except you mostly can't.

Besix2, 2023-08-08:
Also to further discuss this is the pi 4 is able to work as a rubber ducky and stuff but sadly the project is no longer maintained.