RoganDawes/P4wnP1_aloa #313: How can I connect to internet via 4G cat4 card?

zhjygit, 2022-02-19:
I have a rsp 4G cat4 device，and I connect my zero w to 4G cat4 device via usb port (maybe)，my 4G cat4 can connect to internect via SIM card inside。
In common，my zero wh can get IP，for example 192.168.5.8，and connect to internet，I have fininshed it in a common kali linux system。
However，when I user the ALOA kali system，my zero cannot get IP from 4G cat4.
Can someone else test it?
Thank you.
[image]

BDDComputing, 2022-03-07:
We made one like this a year ago and it was working pretty fine, what is your exact issue ?

qkum, 2022-04-07:
> We made one like this a year ago and it was working pretty fine, what is your exact issue ?

He can't get an IP. 

I guess you more exactly ask for the log files showing the precise error messages?