RoganDawes/P4wnP1_aloa #314: P4wnP1_aloa do not boot after restart

kastriotfetahaj, 2022-02-20:
Hi,

I have installed P4wnP1_aloa on Rasberry Pi zero w and the first time when I put the SD card on it works fine.
When I restart it (remove the data cable from the computer and put it back) it will not start. Wifi will not appear and the computer will not recognize it as a USB device.

I have tried different cables and configuration but I am getting the same result all the time.
I have tried the old version of P4wnP1 from this repo (https://github.com/RoganDawes/P4wnP1/) and it works fine.

Can anyone figure out what is problem in my case?

whackyhack, 2022-02-20:
When you say original P4wnP1 works fine, I assume that you mean you installed it on Raspbian or stock Kali, as that one doesn't come with OS image.  Therefore, I suspect that your P4wnP1_aloa image is broken, maybe a problem in the file system or even a card failure.  Try re-image.

kfetahaj, 2022-02-20:
I have flash the latest release in different SD cards, but the same results!

whackyhack, 2022-02-21:
Then HDMI display and USB keyboard is the only means to troubleshoot Kali itself.

qkum, 2022-04-07:
I just installed it myself yesterday and @kastriotfetahaj is correct.

It is buggy.

Sometimes when you reboot it, it will bug and not accept your windows 10 pc connecting to it.

The only fix for that, that I got, for now, is...pulling the power...

- I will likely solder a button on it, activate GPIO, add shutdown with GPIO in /boot/config.txt.

- Now I found out how to connect  to it over bluetooth finally. Now I hope bluetooth can be my backup way of connecting. But who knows, maybe that one also bugs at the same time as the wifi. Time will show. But long term pulling the power will corrupt the SD Card.

qkum, 2022-04-07:
> Then HDMI display and USB keyboard is the only means to troubleshoot Kali itself.

Any device where the wireless connections fail logically enough.

qkum, 2022-04-07:
A fix about the buggy wifi could be many.

`A simple fix could be a while loop that reboots if there have been no devices on the wifo for 5 minutes.`

`A button connected to the GPIO.`

There are many more if you use your imagination.

qkum, 2022-04-07:
The bug is easy to re-create.

* Just click the "reboot" button in the web interface (like 10% of the time)....happened again when testing the web interface over bluetooth...reboot into bluetooth & wifi bugging...